gary-rowe/hid4java #10: Multiple devices with the same vid and pid

mmatloka, 2015-01-11:
Hi,
Thx for making this library, I have encountered a few issues with usb4java ;) I have a question - there can be attached multiple devices with the same vid and pid. How then `hidServices.getHidDevice` will react?

gary-rowe, 2015-01-13:
Sorry for the delay, just seen this message.

I've not tested this scenario, and would rely on the behaviour of the underlying hidapi library. If you are able to construct this as a test case I would be interested in the result.

gary-rowe, 2015-06-03:
Any update on this?

mmatloka, 2015-06-03:
Finally I was not able to test such scenario.

gary-rowe, 2015-06-03:
OK. Thanks for the update. I might be in a position to test this myself now, so I'll leave this open to remind me.

gary-rowe, 2016-03-06:
Finally got around to testing this. Here are my observations:

* Plugging in two devices with the same vendor ID and product ID yielded an enumeration containing both devices. 
* Using a `null` for serial number caused the first device in the enumeration to be attached (a generally random choice). 
* Using a specific serial number only selected that device - be aware that events from the other device may be present and will have to be filtered by discovered serial number

Overall it appears capable of supporting them but this would require comprehensive testing in a production environment to be certain.

Ready for review and close.

andyrozman, 2017-05-16:
Reviewed. Looks ok.

gary-rowe, 2017-06-24:
Merged and closed.

closed 2017-06-24 by gary-rowe

Laivindur, 2020-09-30:
What if the serialNumber is also the same for both devices? So far I know, device SN may or may not be unique. We are working with a provider which produces devices for vets. We have 2 for testing and development purposes and at some point, we realized that both devices had the same vid, pid and iSerial. Don't know if this is normal or we just stumbled upon a mediocre vendor.

The only different serial number we get is the one we get when we address some request to the device and it does respond with the serial number defined by software which is totally different from the one we get from the hardware.

Is there any other device attribute candidate to be UID?

gary-rowe, 2020-09-30:
In USB HID there are three main string descriptors: `product`, `manufacturer` and `serial`. At a low level the `iSerial` value is the index on to a device string which contains the actual serial number. However this index is dereferenced by `Hidapi` into the actual device string so whatever `hid4java` is returning is the actual `serial` string within the device. 

Attempting to dig further with, say, `getIndexedString()` will fail (return code -1) as it isn't fully implemented within `hidapi` (seems to be a TODO item for quite some time).