gary-rowe/hid4java #11: Java 1.5

andyrozman, 2015-01-13:
Hi !

I have application which is built with Java 1.5. Is there specific reason, why Java 7 is used... I have seen that JNA 4 is done with 1.7, is this the only reason? Would it be possible to create 1.5 binaries? I mean I can build them, question is what was reason to use J7?

Take care,
Andy

gary-rowe, 2015-01-13:
The reason for Java 7 was that Java 6 (and earlier) has reached end of life and continued use should not be encouraged. I would urge you to upgrade your JVMs to 1.7 if possible.

Obviously, if that is not possible due to environment restrictions then the code is open source under MIT licence so you should be able to recompile it to a lower byte code level. I've not done this myself but I'd be interested in the outcome so I could add it to the support notes.

andyrozman, 2015-01-14:
When making a library that others can use, it is good to go with oldest version you can (this would be probably 1.5 in your case since Generics are used), so that as many as possible can use it... 
I tried using in on Java 5, but there were some issues:
- there were some @Override tags, overriding methods which you are implementing from interface
- JNA 4 Java version is higher than 1.5 (perhaps 1.6). I tried with last of 3.x versions (3.5.2) and it works.. Did you need any specific functionality from JNA 4, or did you just take the latest when project was started...

gary-rowe, 2015-01-14:
I accept your point about targeting as low a JVM as you can, but there comes a time when backwards compatibility becomes a burden to the main codebase. 

I could remove the **@Override** annotations but it feels wrong to remove that safety net. It's good programming practice to allow the compiler to warn when code isn't actually overriding/implementing a method. This annotation is compatible with Java 6.

When selecting JNA I just went for the latest since it is most likely to have the least bugs and most mature code. 

Overall, I could just about be convinced to lower the minimum JVM to 1.6 but I just can't condone 1.5 on the grounds that it's been EOL since October 2009.

andyrozman, 2015-01-16:
I agree with Java 5 here... If you implement some method (from Interface) you don't override it you implement it...

Even if you support just Java 6, you would need to go with latest JNA 3.x release.

I work for professional company and I know that one some projects we still use Java 5 (custom implementations).

gary-rowe, 2015-01-16:
OK, I'll look into aligning with Java 6 on my next refresh of the project.

gary-rowe, 2015-01-29:
Made various adjustments to support Java6 and JNA 3.5.2.

Ready for review and close.

gary-rowe, 2015-01-31:
Code is now compiled under JDK1.6.

closed 2015-01-31 by gary-rowe