gary-rowe/hid4java #12: Native part loading and missing library

andyrozman, 2015-01-16:
You need to add class for loading native parts from jar... Project nrjavaserial uses something similar and it work great... 
Library for linux x64 is missing at least one version of it, you need to create linux-amd64 directory and copy from linux-x86-64 (leave that one here).

gary-rowe, 2015-01-16:
Thanks for the update. If you're able to offer up a pull request containing these changes it'll help speed things up (I'm maxed out on other work at the moment).

andyrozman, 2015-01-16:
I am sorry... my git knowledge is almost non-existant... I can send you link to correct file.. here it is...
https://github.com/NeuronRobotics/nrjavaserial/blob/master/nrjavaserial/src/main/java/gnu/io/NativeResource.java

You will probably need to modify it a little...

gary-rowe, 2015-01-16:
No worries, thanks for the link.

gary-rowe, 2015-01-29:
I'm not sure of the purpose of the native library loading since JNA already provides this with `Native`. Can you provide more detail? 

In the meantime I've added the Linux support for AMD64.

andyrozman, 2015-01-29:
It's simply for loading native library from jar file... so that you don't need to worry if user has correct file installed. When app is started and library loaded, correct native part will be loaded from jar and you can run with it...

gary-rowe, 2015-01-29:
Is that still within scope of hid4java? If I understand you correctly it should be the case that JNA will load the correct hidapi library from the JAR through `Native` as seen in `HidApiLibrary`.

This mechanism would also be available for any other libraries a user may wish to load since JNA would be available on the downstream application's classpath and thus any other binaries they supplied should be accessible.

Can you give me a use case where this is necessary?

gary-rowe, 2015-01-31:
In order to progress the 0.3.0 release I'll close this issue and if it's still required then a new issue covering it specifically can be raised.

closed 2015-01-31 by gary-rowe