gary-rowe/hid4java #13: Possible incorrect HID packet structure on Windows

gary-rowe, 2015-01-29:
Getting reports of failures on Windows. See [MultiBit HD Issue #379](https://github.com/bitcoin-solutions/multibit-hd/issues/379). It could be due to a HID packet misalignment.

Metin-Ljapo, 2015-01-30:
hi
I had a problem when I tried to build for win 64 bit, the problem was that resource folder is named win32-x86-64 but it was looking for the library under win32-amd64

gary-rowe, 2015-01-30:
Thanks for the update. I'll add an issue to put the win64 library under that directory for the 0.3.0 release. So many architecture variants to list...

gary-rowe, 2015-01-31:
Verified that "The parameter is incorrect" message is removed when the report ID of 0x00 is provided with a change to the length of the report in `HidApi`.

gary-rowe, 2015-02-01:
Verified that this modification works across all platforms.

Closing.

closed 2015-02-01 by gary-rowe