gary-rowe/hid4java #17: Scanning for attached/detached devices

psit, 2015-04-29:
Hi,

two ideas about the scanning for changed devices:
1.) The 500ms intervall cannot be changed when starting the api. It would be great to have that configurable.
2.) The automatic scanning can be stopped but not restarted. I could build my own thread to scan for devices easily (fixing #1 as well), but as you already have that functionality, why not reusing it.

Kind regards,
Peter

gary-rowe, 2015-04-29:
Yes the interval change is a regression. Earlier versions of the code allowed it to be set. The same goes for the device manager life cycle (start/stop). 

I'll schedule in these changes for Release 0.4.0 with the same conditions as I made in #16.

psit, 2015-06-01:
Hi Gary,

I created a pull request for the issue.
https://github.com/gary-rowe/hid4java/pull/21

Additionally this update supports disabling the shutdown hook (as this interferes with my application, where your shutdown hook might hinder my own hooks).

gary-rowe, 2015-06-01:
Thanks, psit - this is much appreciated. I'm a bit tied up today, but I'll review the PR tomorrow and see about merging it in.

gary-rowe, 2015-06-02:
Merged, so closing.

closed 2015-06-02 by gary-rowe