gary-rowe/hid4java #19: Can't read port in Windows 8.1

EdisonDelgado, 2015-05-27:
Hi, thank so much for the Api is very useful for me.Actually i'm working in a  device that have a set of external sensors like Temperature, Humidity, Light ,etc.  It´s device report an array of byte when it is writing a command every x interval.

My trouble is only in Windows can't read the response from my device. 

This method write a command in to device, REPORT_ID = (byte) 0
```
    public int write(byte command) {
        byte[] data = new byte[1];     
        data[0] = command;
        int result = currentDevice.write(data, data.length, REPORT_ID);
        return result;
    }
```
This try to read port every 100 ms.
```
    public void report() {
        timer.schedule(new TimerTask() {
            public void run() {
                byte inBuffer[] = new byte[PACKET_LENGHT];
                int val = currentDevice.read(inBuffer);
                switch (val) {
                    case -1:
                        //System.err.println(currentDevice.getLastErrorMessage()); always throw null
                        break;
                    case 0:
                        break;
                    default:
                        if (inBuffer != null) {
                            reportData(inBuffer);
                        }
                        break;
                }
            }
        }, 0, 100);
    }
```

I tested the same code in Mac and worked perfectly.

What am I doing wrong?

Thank in Advance.

gary-rowe, 2015-05-28:
You might be falling foul of a quirk in Windows when writing report IDs of 0x00. Have a look at `HidApi.write()` around line 369 with a debugger and see if it is affecting your situation.

EdisonDelgado, 2015-05-28:
I wrote my command with the following structure and worked without problems.
```
    public int write(byte command) {
         byte[] data;
        if(PlatformUtils.isWindows()){
            data = new byte[2];
            data[0] = (byte) 0;
            data[1] =  command;
        }
        else{
            data = new byte[1];
            data[0] = command;
        }
        return currentDevice.write(data, data.length, REPORT_ID);
    }
```
You write the message with different structure for Windows. My device doesn't work because the command becomes endpoint.

Thank you!!

closed 2015-05-28 by EdisonDelgado