gary-rowe/hid4java #22: unable to open device

NathanSweet, 2015-06-13:
I see my device when using `getAttachedHidDevices` but it doesn't open:
```
for (HidDevice hidDevice : hidServices.getAttachedHidDevices()) {
	if (hidDevice.isVidPidSerial(0x077d, 0x0410, null)) {
		System.out.println(hidDevice);
		System.out.println(hidDevice.open());
	}
}
```

Output:
```
HidDevice [path=\\?\hid#vid_077d&pid_0410#7&325c568b&1&0000#{4d1e55b2-f16f-11cf-
88cb-001111000030}, vendorId=0x77d, productId=0x410, serialNumber=?, releaseNumb
er=0x400, manufacturer=Griffin Technology, Inc., product=Griffin PowerMate, usag
ePage=0xc, usage=0x1, interfaceNumber=-1]
false
```

I'm at a loss. Any ideas?

gary-rowe, 2015-06-13:
Interesting. Your device is declaring itself as a Consumer Audio Control (usage page 0x0C, usage ID 0x01) which is opened by Windows for System use. Here's [an article about it](https://msdn.microsoft.com/en-us/library/windows/hardware/ff543477%28v=vs.85%29.aspx). 

However the access mode from the system is Shared indicating that it should be possible to open it yourself. Unfortunately, given that it's listed there means that it may not be possible to use `hidapi.dll` (the underlying HID API that hid4java wraps) and you may need to explore the more direct usb4java alternative.

NathanSweet, 2015-06-14:
Thanks for the help. Man, what a pain this has been! usb4java was a failure, it just didn't not work right. Eventually I switched to LibUsbJava and finished my project:
https://github.com/EsotericSoftware/powermate
Here is where the magic happens:
https://github.com/EsotericSoftware/powermate/blob/master/src/powermate/PowerMate.java#L28
Cheers!

gary-rowe, 2015-06-14:
That's good to know and I'm glad you were able to resolve the issue. [LibUsbJava](http://libusbjava.sourceforge.net/wp/) is quite an old project (2008 was last release) but at least it worked so no complaints there.

closed 2015-06-14 by gary-rowe