gary-rowe/hid4java #24: how to add hid4java to my project?

mohamnexus, 2015-06-27:
Hi
I work on java project in netbeans and i want use usb hid,how to add hid4java to my project?
can i use jar file of hid4java for library?
please learn me step by step for add your library to my project.
please help me.

gary-rowe, 2015-06-28:
Hi @mohamnexus. Thanks for your interest in hid4java. Can you verify that you've read the README and installation instructions? If your project uses Maven it will be very straightforward. If not then you can use one of the pre-built releases here: https://github.com/gary-rowe/hid4java/releases

mohamnexus, 2015-07-29:
Hi
I could use hid4java in my project and i can send data to usb,but i can't get data from hid usb.
i want use getReport() method , please help me.
thanks

gary-rowe, 2015-07-29:
Can you provide a code sample indicating what you're trying to do? There is extensive documentation on the API detailing how to use it.

mohamnexus, 2015-07-30:
[image]
***********************************************************************
and code is:
/*
 * The MIT License (MIT)
 *
 * Copyright (c) 2014-2015 Gary Rowe
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 * SOFTWARE.
 *
 */

import java.io.IOException;
import org.hid4java.*;
import org.hid4java.event.HidServicesEvent;

import java.util.concurrent.TimeUnit;

import static java.util.concurrent.TimeUnit.NANOSECONDS;

/**
 * <p>
 * Demonstrate the USB HID interface using a production Bitcoin Trezor</p>
 *
 * @since 0.0.1  
 */
public class UsbHidTrezorV1Example implements HidServicesListener {

    static final int PACKET_LENGTH = 8;
    private HidServices hidServices;
    int vid = 0x471;
    int pid = 0x2201;

    public static void main(String[] args) throws HidException {

        UsbHidTrezorV1Example example = new UsbHidTrezorV1Example();
        example.executeExample();

    }

    public void executeExample() throws HidException {

        System.out.println("Loading hidapi...");

        // Get HID services
        hidServices = HidManager.getHidServices();
        hidServices.addHidServicesListener(this);

        System.out.println("Enumerating attached devices...");

        // Provide a list of attached devices
        for (HidDevice hidDevice : hidServices.getAttachedHidDevices()) {
            System.out.println(hidDevice);
        }
[... 108 lines cut ...]
                    remainingNanos = end - System.nanoTime();
                }
            }
        } finally {
            if (interrupted) {
                Thread.currentThread().interrupt();
            }
        }
    }

}
******************************************************
and output is:

[image]

******************************************************
and my device is Hid low data-rate devices.

now i can send data to device but i can't get data from hid usb.
i want getreport()  for get data.
my device (usb) have no problem.

please help me and if you can't help me show me other library for this work.
thanks.

mohamnexus, 2015-08-04:
Hi
Ignore my previous question.
now i can send data to USB with   trezor.write(message, PACKET_LENGTH, (byte)  0);
and i can get feature report with   trezor.getFeatureReport(data, (byte)  0 );
but when change report ID to 0x1 i can send data but i can't get data from usb.
why?

gary-rowe, 2015-08-04:
When you change the report ID in USB then it changes the function of the device. You need to consult the documentation that comes with it in order to correctly determine which reports are supported and what they do.

mohamnexus, 2015-08-05:
Thanks for comment.
I have an application for usb device test (SimpleHIDWrite) that this app supporting getFeatureReport with 0x01 report id but i can't use getFeatureReport with hid4java library.
help me please.
[image]

but in hid4java
[image]

gary-rowe, 2015-08-05:
Judging by the `ArrayIndexOutOfBoundsException` and the structure of the `getFeatureReport` method it seems that the `hid_get_feature_report` native method is either returning a value of `0` or breaking its contract and returning more than `data.length + 1`. 

You'll need to run your code in debug mode and examine the value of `res` at line 309 to establish which it is. The source code of hid4java is available through the standard Maven build process but you may need to activate this feature through NetBeans.

gary-rowe, 2015-08-23:
This bug has been fixed in release 0.4.0 so closing here

closed 2015-08-23 by gary-rowe

frankmeho, 2024-04-08:
Hi @gary-rowe ,
As far as I know, GetReport mentioned by @mohamnexus is different from getFeatureReport.
I've used the latest hid4java-develop with develop-SNAPSHOT version and it works well on most of my handled projects using the USB HID interface to send/receive commands.
However, there're some specific platforms that require sending SetReport first and then sending another GetReport to obtain its results. It seems that the current hid4java-develop does not provide this GetReport API yet.

frankmeho, 2024-12-31:
Hi @gary-rowe ,

I take a look again about the discussion between you and @mohamnexus , look like the getReport seems to be solved with hid4java-0.4.0, then I will try it and let you all know the result.

Thanks.
Sincerely,
frankmeho

frankmeho, 2025-01-02:
Hi @mohamnexus , can you provide me the hid4java-0.4.0 of develop branch which can fix the issue of GetReport?
I've tried the current hid4java from the web site of https://mvnrepository.com/artifact/org.hid4java/hid4java/0.4.0, but so far I cannot find a workable one.