gary-rowe/hid4java #25: Include a proper license header in source files

hegjon, 2015-07-15:
It would make it easier to redistribute hid4java if each source file contains a license header.

Check https://bugzilla.redhat.com/show_bug.cgi?id=1230949#c8 for details.

gary-rowe, 2015-07-15:
Hi Jonny - thanks for continuing with this. I've added this to the 0.4.0 release which I expect will get released with the [0.1.3 release](https://github.com/bitcoin-solutions/multibit-hd/milestones/Release%200.1.3) of MultiBit HD.

hegjon, 2015-07-15:
Thanks, glad to hear :)

gary-rowe, 2015-07-24:
Added the copyright notices as required.

Ready for review and close (@hegjon can you do this when you're available please?)

hegjon, 2015-07-24:
Looks good

closed 2015-07-24 by hegjon

hegjon, 2015-07-25:
Here is the output from fedora-review license check:
````
MIT/X11 (BSD like)
------------------
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/UsbHidTrezorV1Example.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/HidDevice.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/HidDeviceManager.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/HidException.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/HidManager.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/HidServices.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/HidServicesListener.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/event/HidServicesEvent.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/event/HidServicesListenerList.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/jna/HidApi.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/jna/HidApiLibrary.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/jna/HidDeviceInfoStructure.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/jna/HidDeviceStructure.java
hid4java-aa188231cc912041125816e6343be5dd09568040/src/main/java/org/hid4java/jna/WideStringBuffer.java
````

gary-rowe, 2015-07-25:
Thanks, looks like the Intellij Copyright tool works :-)