gary-rowe/hid4java #27: Support for ARM

MaxRoma, 2015-08-24:
Hi!
I am working on embedded application that is running on ARM Linux. Is there any instruction on how to enable hid4java on ARM environment? I was able to find manual only for javahidapi http://www.finchrobot.com/compiling-javahidapi-raspberry-pi?page=9. Am my correct that https://github.com/cibuddy/cibuddy/blob/master/drivers/hid/src/main/resources/lib/linux/arm7/libhidapi-jni.so file has to be saved under linux-arm/libhidapi.so inside hid4java-0.4.0.jar. And if so, is it possible to add pre-compiled library in hid4java repository for everyone else to use?

Take care,
Max

gary-rowe, 2015-08-24:
Yes. hid4java uses JNA under the covers and it [defines the naming convention](https://github.com/java-native-access/jna/blob/master/build.xml) to use. You'll note from that file that `linux-arm` would be appropriate. 

I've just pushed a more recent version of `libhidapi` on the `develop` branch based on an RPi compile. Can you clone and build locally (`mvn clean install`) and then use `hid4java-develop-SNAPSHOT.jar` in your project. Ideally it'll work out of the box.

If it doesn't then I'll see about running up a VirtualBox or similar to emulate a suitable ARM environment to expand the library but that may take some time due to other projects requiring attention.

MaxRoma, 2015-08-25:
Hi!

Thank you for quick reply, you've saved my live) I've tested DEV version and it worked. Well at first it was complaining that there is no **libudev.so.0** file *(libudev.so.0 was replaced with libudev.so.1 from ubuntu 14.04 and forward)*. Nevertheless after applying this command **sudo ln -sf /lib/arm-linux-gnueabihf/libudev.so.1 /lib/arm-linux-gnueabihf/libudev.so.0** everything started to work like charm on ARM.

closed 2015-08-25 by MaxRoma

gary-rowe, 2015-08-25:
No worries - glad I could help. This fix will go out in the 0.5.0 release which will probably be in late September judging by the downstream project driving development - [MultiBit HD](https://multibit.org).

I'll update the dev notes to reflect the symlink - good catch.