gary-rowe/hid4java #28: Classes in root package

jkropf, 2015-09-16:
Do not add classes to the default (root) package, this is bad style and causes errors using maven to build projects.
Currently the class UsbHidTrezorV1Example.class is located in the root package.

You can no longer import from default packages. (You used to be able to in Sun's early javac implementations, but this was not allowed for in the language spec.)

gary-rowe, 2015-09-16:
Clearly this class is only intended for demonstration purposes and should not be referenced in actual use. It is placed there to simplify accessing it outside of an IDE through command line Maven.

That said, your point is well taken. Perhaps the library is mature enough now to move this under src/test/java and into its own example package.

I'll add this to 0.5.0.

jkropf, 2015-09-16:
Nice, thanks. Otherwise one can't build an application using maven. My
workaround was to delete it from the jar.

Kind regards,
Johannes

gary-rowe, 2016-03-06:
Moved the example around and updated the README as required.

Ready for review and close.

andyrozman, 2017-05-16:
Either I am looking in at wrong branch or we still have some references to TrezorExample in wrong location in README. Could you please check that, so that I can finish review?

gary-rowe, 2017-06-24:
Verified the link and Maven execution. Ready to merge.

gary-rowe, 2017-06-24:
Merged and closed.

closed 2017-06-24 by gary-rowe