gary-rowe/hid4java #30: HidDevice.onInputReport not supported

karunadheera, 2015-09-20:
This is more kind of a feature than a bug!

I found https://github.com/nyholku/purejavahidapi to support above feature among others. However purejavahidapi library does not support listeners on device addition (only removal is supported) - so I am not going to keep using that library only if hid4java library is going to support arbitrary retrieval of data from HID devices.

HID devices can send arbitrary data on certain events (such as key presses) and hid4java as of now does not support this. Could anybody look into implementing this feature.

Thanks in Advance!

gary-rowe, 2015-09-20:
Thanks for letting me know. I'll add it to the 0.5.0 release milestone and take a look when I get some spare time.

karunadheera, 2015-09-26:
Thanks @gary-rowe! Looking forward to 0.50 milestone. Have a nice day!

gary-rowe, 2016-03-06:
I may have some time to look at this shortly. Can I verify that you're looking for is essentially a poll on the input report buffer followed by a dedicated event containing the data after it is fully read?

leif81, 2016-03-09:
@gary-rowe That's how understand @karunadheera. It sounds very useful. I had a similar thought when I started using hid4java. I suspect most users of the library end up implementing the same polling logic in a thread in their application code. It'd make things a little nicer to use if the hid4java library provided a callback listener interface like _purejavahidapi_.

retromuz, 2016-05-31:
That's right @gary-rowe.

gary-rowe, 2020-07-01:
I doubt I'll be able to implement this any time soon so I'll close the issue.

closed 2020-07-01 by gary-rowe