gary-rowe/hid4java #31: problem with binary

andyrozman, 2015-09-22:
Hi !

I used 0.3.1 for testing so far, now my library (classes) is ready for production, so I build 0.4.0 version locally, but now when I try to run it, I get problem with binary file (I use linux-amd64). This is what I get...

symbol lookup error: /home/andy/workspaces/andy-iplayground4/ggc/hid4java/target/classes/linux-amd64/libhidapi.so: undefined symbol: libusb_init

Do I need any special dependencies on system to run with new version?

And before you ask, I build my own version because I need 1.5 compatibility.

Andy

gary-rowe, 2015-09-22:
This could be due to the `libusb` library being renamed on your system, possibly to include the version number. You can test this by means of `ls /usr/lib/libusb*.a` and seeing if `/usr/lib/libusb-1.0.a` or similar is present in the result.

You might be able to use the instructions in [this Stack Overflow question](http://stackoverflow.com/q/9558909/396747) to work out what you need to do on your system to overcome this situation. One answer suggests using `LD_PRELOAD`.

andyrozman, 2015-09-22:
But why would 0.3.1 work and 0.4.0 not... When I use official version of 0.3.1 it works, when I use 0.4.0 I rebuilt not...
Actually I don't have any files like this.

andyrozman, 2015-09-22:
I tried using 0.3.1 binaries with 0.4.0 java sources. This one works better, but read command doesn't work. I have seen that there were some changes (HidDeviceInfo file was removed), which might be reason why read command doesnt work (I get -2 returned instead of data)...

Did you change how you make binaries from 0.3.1 till next (current) version?

gary-rowe, 2015-09-22:
Hmm, looking over 0.3.1 there's nothing in the `pom.xml` to indicate a change. Both Java 6 and JNA 4.1.0 were in place by that time (0.3.0 required Java 7+ but this was later relaxed).

If you got the binaries from the Bitcoin Solutions staging repo (as referenced in the README) then it's possible that later ones were compiled under Java 8 targeting a Java 6 language level and byte code format.

From the look of the error message it's a problem with the dynamic linking between the signal11/hidapi library and your libusb. I've no idea why earlier binaries were working unless something changed on your system between when you compiled them and now. Perhaps it would be worth re-compiling the old ones and seeing if they still work?

andyrozman, 2015-09-22:
I didn't compile binaries. I just took them from your jar file. Now I took them from your repository... they are stored under src/main/resources.
Situation really seems weird... I will try to get old sources and recompile them and see if it works...

andyrozman, 2015-11-11:
I recompiled with old sources  0.3.1  (and using also old binaries) and it works now (not with new ones, it seems like you might be correct, perhaps new binaries are targeting higher java version)... Since I need this for java 5 I made fork of this project, which is now "locked" at 0.3.1. I will merge changes if you make some, but since you also changed Java implementation from 0.3.1 to 0.4 I will have to be very carefull... 

If someone needs Java 5 version it has name hid4java-j15 (here on github)...

You can close this ticket... Thanks for your help and you did good work with this library... Start of using USB hid in my project was much easier than I expected.

gary-rowe, 2015-11-11:
OK. Thanks for the feedback. Closing.

closed 2015-11-11 by gary-rowe