gary-rowe/hid4java #33: Running in Linux (Ubuntu 15/10)

xgonc, 2015-11-11:
Hello,

I built my application to a single JAR file, and it works as expected in Windows 7/8, and Mac (mountain lion). However in Ubuntu 15/10 I see the devices with lsusb, but I cannot get them from the Java code. I suspect this a dependency issue. 
Libusb is installed in a non standard location, so I created a symbolic link like this:
/usr/lib/libusb.a -> /usr/lib/i386-linux-gnu/libusb.a

I don't have any more clues to solve this...

Note java installed:
java version "1.7.0_85"
OpenJDK Runtime Environment (IcedTea 2.6.1) (7u85-2.6.1-5)
OpenJDK Server VM (build 24.85-b03, mixed mode)

Please help me. Thanks.

andyrozman, 2015-11-11:
You need to set udev rule for your device, or else your user has no access to devices. In ubuntu you are running as non-privileged user, access to hardware needs to be permitted... Google "udev rules ubuntu"

gary-rowe, 2015-11-12:
Thanks @andyrozman. The [README](https://github.com/gary-rowe/hid4java/blob/develop/README.md) file provides the details you need @xgonc.

closed 2015-11-12 by gary-rowe

xgonc, 2015-11-12:
That was the issue. Solved. Thanks.