gary-rowe/hid4java #34: Attempting to run sample program errors out (Windows 10)

AbbottWC, 2015-11-12:
I'm switching over from Java HIDAPI to HID4Java.
I am running JNA 3.5.2.

The exception message I'm getting is : 

Exception in thread "main" org.hid4java.HidException: Hidapi did not initialise: Unable to load library 'hidapi': JNA native support (win32-x86/hidapi.dll) not found in resource path (C:\Users\Will\Documents\NetBeansProjects\Libraries\jna-3.5.2.jar;C:\Users\Will\Documents\NetBeansProjects\Libraries\platform-3.5.2.jar;

Likely it means that I'm doing something wrong but... I don't know what it is.

I'm running Windows 10 (x64) on a 32 bit JVM (for logistical reasons - some people run dinosaurs).

Please, help?

andyrozman, 2015-11-12:
This will not work. You need at least 4.1 version of JNA. JNA 3.5.1 doesn't automatically load natives, 4.1 does. You will have to upgrade your local JNA to at least 4.1.0 (this one works good). Are you limited with Java version too? You need to have at least Java 6 to run this.

AbbottWC, 2015-11-12:
Okay thanks. There's one problem though - another project I'm using (LibVLC Java) requires JNA 3.5.2. I don't think that it's possible to run both, is it?

AbbottWC, 2015-11-13:
One other thing - I do not know if it matters, but I'm developing with the Netbeans IDE.

AbbottWC, 2015-11-13:
Okay I goofed... I downloaded the entire project as a .zip file and tried to make it work as a Java project instead of... whatever it is.

I've never worked with... whatever this is, but when I tried to debug it as is (straight out of the .zip), it worked fine... so I'm sorry for taking your time.

closed 2015-11-13 by AbbottWC

gary-rowe, 2015-11-13:
Hi @AbbottWC. The hid4java project is a Maven project which is how many Java projects are created these days. There's [more information in the wiki](https://github.com/gary-rowe/hid4java/wiki/How-to-build-the-project) that may help you become more familiar with the process.