gary-rowe/hid4java #35: HidServices.getHidDevice( ... ) returns NULL when device is attached on Windows 10

AbbottWC, 2015-11-13:
I've built the Maven project and extracted the resultant JAR file to use as a reference in a test project.

I have the following code snippet : 

    public static boolean Initialize() throws HidException{
		if (JR3Controller.Ctor != null) return true;
		JR3Controller.Ctor = new JR3Controller();
		JR3Controller.Ctor.mgr.addHidServicesListener(JR3Controller.Ctor);
		return JR3Controller.Ctor.dev != null;
    }
	
	private JR3Controller(){
		this.mgr = HidManager.getHidServices();
		this.dev = this.mgr.getHidDevice(0x0003, 0x1001, null);
		if (this.dev != null){ /*Do Stuff; Doesn't matter, dev is always null*/ }

Following the code execution, I can see that this.mgr is being set to a non-null value, but when I try to open my device, well, it fails...

This was the same issue I was running into using the Java HIDAPI library (which, coincidentally, is why I switched over to hid4java).

Something I read somewhere said that Windows 10 hijacks devices that are connected to the system as soon as they are connected (My knowledge of such things is nil, however); could this be the problem here? Am I doing something incorrectly?

andyrozman, 2015-11-14:
When using windows this can always happen... I am using hid4java in my project. I use it on linux and one of other users is using it on windows (7). For device we are using there is no problem, but we are using device which is combined (Storage device with HID interface). So since device was detected as Storage USB, we had no problems connecting to it. But if you have device that requires special driver and exclusive access, then you are right. Windows might hijack it... 

You could theoretically install libusb and filter on this device, but I am not sure if it would be any better for hid4java... Perhaps owner of project can say what happens if libusb is detected on windows system...

gary-rowe, 2015-11-14:
Thanks @andyrozman. Also #18 contains some further useful information on device connectivity.

gary-rowe, 2016-03-06:
Just to update this. I've seen this code working on Windows 10, but a recent push as part of #36 may be relevant.

Can you verify if you are experiencing this problem with the latest code on the `develop` branch as of today?

gary-rowe, 2017-06-24:
Closing due to no activity.

closed 2017-06-24 by gary-rowe