gary-rowe/hid4java #39: Device scanning interferes with keyboard key presses

leif81, 2016-03-24:
Ever since I integrated *hid4java* into my project I've noticed that when my application is running my USB keyboard behaves oddly. Specifically what happens is that when I'm typing on my keyboard (in another application) the occasional keyboard key press gets duplicated. So if I intended to type **hello world** the result may actually be **helloo world** (notice the duplicate *oo*). Exiting my application that is using *hid4java* resolves the issue and future key presses are accurate.

Today I decided to investigate this problem a bit further and I think I've narrowed it down to something that may be helpful. I believe the issue is related to the device scanning. Because if I disable device scanning by passing a value of `0` to the `scanInterval` of the `HidServices` constructor the problem goes away.

I'm observing this issue when running my application on Windows 7 64-bit. I also run my application on a Raspberry Pi, but I haven't noticed the same problem there.

andyrozman, 2016-03-24:
With current implementation this is little impossible to do, especially if you use HidManager. We would probably need additional method, which could pass this scan interval to HidServices instance. Or perhaps scan interval could be static variable (which we could set before calling HidManager).

andyrozman, 2016-03-25:
I made few changes to code, which could help with this problem. @gary-rowe could you look at pull request and merge it?

leif81, 2016-03-26:
Thanks @andyrozman . This will make it a little nicer to access the singleton to avoid the issue. 

But, any idea why the scanning is interfering with key presses? It would be nice to be able to have scanning on Windows.

gary-rowe, 2016-03-26:
Thanks @andyrozman for the PR. I'll be able to dedicate time to this early next week.

seanidzenga, 2017-05-24:
Has there been any progress on this issue?

gary-rowe, 2020-07-01:
Apologies for the slow response. There has been a lot of work in the scanning code so I think this has been resolved. If you'd like to try again, please use latest develop-SNAPSHOT version.

closed 2020-07-01 by gary-rowe