gary-rowe/hid4java #41: 1-byte Reports do not get written correctly

StefanKrupop, 2016-05-18:
I think there is an off-by-one error in the "write" method of org.hid4java.jna.HidApi:

The JavaDoc says the Report data should not include the Report ID. It will get prepended to the data automatically.
However, both the Windows- and regular code path execute the arraycopy only if the data length is larger than 1. Thus, a 1-byte Report gets written out, but not including the supplied data...
I think it should be "if (len >= 1) {".

Kind regards,
Stefan

gary-rowe, 2016-05-18:
Can you verify that this still occurs with the develop branch version of hid4java?

StefanKrupop, 2016-05-18:
I did not (yet) check it with a compiled version of the branch, but the section I meant is also in the development version:
https://github.com/gary-rowe/hid4java/blob/c08536c43bd3ea9e32e82941c36ebc02324df985/src/main/java/org/hid4java/jna/HidApi.java#L432-L439

When len=1 the arraycopy in both paths will not get executed.

marek-trmac, 2016-08-22:
I found the same problem, and I verified that fixed condition `if (len >= 1)` works.

gary-rowe, 2017-06-24:
Fixed in pull request.

closed 2017-06-24 by gary-rowe