gary-rowe/hid4java #44: Device not initialized

VmKid, 2016-07-24:
I am attempting to open a device (A Buzz! controller) to be written to. Unfortunately, the writes don't seem to be working, Upon printing the last error message, it tells me "Device not Initialized". The String representation of the device is like this:

`HidDevice [path=\\?\hid#vid_054c&pid_0002#6&1795862b&2&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}, vendorId=0x54c, productId=0x2, serialNumber=null, releaseNumber=0x1101, manufacturer=Logitech, product=Logitech Buzz(tm) Controller V1, usagePage=0x1, usage=0x4, interfaceNumber=-1]`

Any help would be appreciated.

andyrozman, 2016-07-24:
Did you call open on device after you 'found' it. I had problem like that.
I got device from manager and than just tried to use it without opening it
(opening was not be needed on version 0.3.1 of library).
Andy

VmKid, 2016-07-24:
Yes, I did. The code that I'm using in this case is below:
```
if(controllerDevice != null){
            if(!controllerDevice.isOpen()) controllerDevice.open();
            System.out.println(controllerDevice.getLastErrorMessage());
            controllerDevice.write(message, MESSAGE_LENGTH, (byte)0);  
}
else System.out.println("Couldn't change lighting!");
```

gary-rowe, 2019-10-29:
Closing as this is likely covered in #81.

closed 2019-10-29 by gary-rowe