gary-rowe/hid4java #47: hid write not sending correct number of bytes

bphagan, 2016-08-29:
I just started working on a project to communicate with a hid device.
Since the Java app needs to run on Windows and Linux(maybe Mac down the road), I thought hid4java would be a good fit.
I was able to get everything working and I can connect to my device.
The issue I have and can't figure out is:
When I send a 16 byte packet plus 1 byte report ID set to 0x01, the return bytes written length is 1025.
I can't seem to understand why I am getting 1025 vs 17. This is causing my hid read to be too big and after the initial few bytes, wrong data.
I am running on Windows 7 x64. 

To be more specific, I put a print statement right after this call in HidApi:
int res = hidApiLibrary.hid_write(device.ptr(), report, report.buffer.length);
res is equal to 1025 after the call.
report.buffer.length was equal to 17 right before the call.

Any idea why I may be getting a different return value?

Thanks,
Brett

gary-rowe, 2016-08-29:
It might be an artifact of an off-by-one error. Can you try using the develop branch of the library, and try setting `HidApi.dropReportIdZero = true;` before opening a device on Windows 7 only. There is a bug which could cause a report ID to be included in the packet that gets sent that could be confusing the device into returning different values.

bphagan, 2016-08-29:
To clarify, I am using Report ID = 0x01 which the device is expecting.
The part that is not behaving is the return values from hid write and hid read.
I have length set to 17 which includes the ReportID. 
The value returned from the hid write is 1025.
When I perform a 2 hid reads, I get the data expected in both for the command sent in write.
So, it appears I have a working system, but was thrown off by the write return value. 

Thanks.

closed 2016-09-06 by bphagan

gary-rowe, 2016-09-07:
Did you find out what the problem was? I couldn't offer up any more help since I didn't have a means to replicate your environment.