gary-rowe/hid4java #48: Creating Runnable jar file

bphagan, 2016-09-06:
I have everything working so far within Eclipse project. 
I exported to create a Runnable jar file. 
The gui comes up but no usb connection.
When I run from cmd line with console enabled, I get a resource not found error.
See console text below.
I tried exporting several ways but similar issue.
When I open the jar file, I see a resources\win32-x86-64 folder with the hidapi.dll in it.
I am sure it is something simple I am overlooking, but I can't find it.

Manifest
Manifest-Version: 1.0
Class-Path: .
Main-Class: mxUSB.MXBootTool

"
C:\Users\r12528\Desktop>java -jar iMX_Boot_Utility.jar
Launching application...
Operating System : Windows 7 6.1
Java Version: 1.8.0_91
Java Home: C:\Program Files\Java\jre1.8.0_91
Showing GUI.
Loading hidapi...
Exception in thread "AWT-EventQueue-0" org.hid4java.HidException: Hidapi did not
 initialise: Unable to load library 'hidapi': Native library (win32-x86-64/hidap
i.dll) not found in resource path ([file:/C:/Users/r12528/Desktop/iMX_Boot_Utili
ty.jar])
        at org.hid4java.HidDeviceManager.<init>(HidDeviceManager.java:89)
        at org.hid4java.HidServices.<init>(HidServices.java:88)
        at org.hid4java.HidServices.<init>(HidServices.java:76)
        at org.hid4java.HidServices.<init>(HidServices.java:63)
        at org.hid4java.HidManager.getHidServices(HidManager.java:50)
        at mxUSB.MXConnectUSB.<init>(MXConnectUSB.java:57)
        at mxUSB.MXBootTool.createAndShowGUI(MXBootTool.java:92)
        at mxUSB.MXBootTool$1.run(MXBootTool.java:85)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionP
rivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

gary-rowe, 2016-09-07:
I think you have a mangled project build process. Normally a self-executing JAR is created using the Maven Shard plugin and it should just pull in the hid4java JAR from Maven Central. Can you explain a little more how you're configuring Eclipse?

bphagan, 2016-09-07:
I was trying extract require libraries and package required libraries options.
I was able to get the package required libraries option to work by manually adding resources/ to the MANIFEST.MF file like this:
Rsrc-Class-Path: ./ jna-4.1.0.jar resources/

Now it works as expected on Windows and Mac OS X.

But on Linux, I need to run in terminal with sudo. IS there a trick to use this without sudo?

gary-rowe, 2016-09-07:
OK, glad you got it working in your environment. The `README.md` contains useful information for getting the library to work on Linux. In particular, you'll probably want to refer to this section:
https://github.com/gary-rowe/hid4java#my-device-doesnt-work-on-linux

bphagan, 2016-09-07:
Thanks Gary. I tried that but it didn't seem to help. 
No problem since most of the Linux users are familiar enough with terminal that they can just run it with sudo.

closed 2017-06-24 by gary-rowe