gary-rowe/hid4java #49: Problems with hid.read callback on OSx

lorenberg, 2016-09-13:
Hello Gary,

after a few months, here I am again asking for your help.
I was finally able to use your library for transmitting some data from our MAC to our HID device.
The transmission direction (hidDevice.write) works flawlessly.
Due to a redesign of the application, we wanted to add the reverse channel, namely we want the device to return some data when the control software asks to do so. 
Well, this works fine on Windows, but on OSx the application blocks when calling the hidDevice.read() instruction. I am trying to understand what happens, but there is no exception thrown and no debug message. The application simply freezes (it does not even shutdown).

Do you have any suggestion on how to find the error?
In my code, the read() instruction is called after the write() (I will send a command code to the device, telling it that I need some data, and it should reply with those data). 
Again, this works fine in Windows.

Thanks for your precious support,

Best

Lorenzo

lorenberg, 2016-09-15:
If it could be of help, I debugged up to the hidApiLibrary.hid_read(...) method, and here it fails.
So, my guess is that there is some issue on the -dylib library....
Any help would be great :)

gary-rowe, 2020-07-01:
Hi. Apologies for the slow response. I think this has been resolved by the latest develop-SNAPSHOT which uses the latest `hidapi`.

closed 2020-07-01 by gary-rowe