gary-rowe/hid4java #50: Reopen device causes null device

begonaalvarezd, 2016-09-29:
Hello, I already wrote my issue [here](https://github.com/gary-rowe/hid4java/issues/32#issuecomment-247973029) but I think it may be better to open it as a new one.
Hello! I am writing you because I have been days struggling with an error. I am experiencing a similar issue as @xgonc . There is one special functionality (all others work cross-platform but this one is not working on windows 7) that I have developed to update the firmware of a control board with which I am communicating with using your library. 
To be able to access the flashing mode of the board, I have to disconnect first from the board in normal mode (which has a productId = 0x0001 for example), and access the bootloader mode that lasts 2 seconds after the board is restarted, being this mode with productId = 0x1001. To do so, I send a restart command to the board, wait, and try to connect to the bootloader mode. This is working well under Ubuntu 14, Windows 8.1 and Windows 10 so far, but under Windows 7 it always throw the same exception:

`java.io.IOException: Cannot read from USB device! `

So, the error jumps in when I never unplug the USB of the board but just send a restart command and try to connect ,BUT if I unplug and plug it again and try to flash the board withing those 2 seconds that the mode is open...it works!

This error seems very weird to me because after deep debugging for days I have managed to see that when trying to connect to this bootloader mode after the restart command, which involves a disconnection of previous mode and re-connection to another mode (for the program a new USB device), the detection actually works but is the read() which fails returning  getLastErrorMessage(): "device is not connected". 

When trying to communicate I first do: 

```
device = hidServices.getHidDevice(VENDOR_ID, deviceMode.getValue(), null);

if (device != null)
{
	System.out.println("CONNECTED");
}
```

Which prints CONNECTED on the bootloader mode!

But then, deviceRead() throws `java.io.IOException: Cannot read from USB device! `

```
public byte[] deviceRead() throws IOException
	{
		final byte[] data = new byte[200];
		int read = 0;

		read = this.device.read(data, 250);

		/* Error occurred - nothing read */
		if (read < 0){
			throw new IOException("Cannot read from USB device!");
			//read = 0;
			//System.err.println("USBHID LAST ERROR MESSAGE: " + device.getLastErrorMessage());
		}

		byte[] ret_val = new byte[read];
		for (int i = 0; i < read; i++)
			ret_val[i] = data[i];

		return ret_val;
	}
```

Due to 

```
public static HidDeviceStructure open(String path) {
    Pointer p = hidApiLibrary.hid_open_path(path);
    return (p == null ? null : new HidDeviceStructure(p));
  }
```

Is returning ` P = null `. 

I have already tried re-initializing the library when disconnecting, using single instances of the library objects as some other issues here recommend...everything with no luck.

Any help/idea/holy-light would be really really helpful. I am breaking my mind here trying to understand why is not working and I am pretty sure I am missing something important. 

Have a great day!

begonaalvarezd, 2016-10-05:
Hello again! 

After many days debugging it resulted to be a bug in the controlboard itself and not in the library or any missuses. Now everything is working smoothly and crossplatform, as usual.

Thank you for this great library!

Best :)

closed 2016-10-05 by begonaalvarezd

gary-rowe, 2016-10-05:
Thanks for getting back to me on this and glad you were able to isolate the bug. 

Hope the project goes well :-)