gary-rowe/hid4java #52: scales read once then never again

rwu2359, 2017-02-28:
I have some simple Groovy to read from DYMO M10 scales.

It works the first time when run, but then the device is never found again until a complete shutdown of the environment
Any clues???

import org.hid4java.*
import org.hid4java.jna.*

Short VENDOR_ID = 0x0922
Short PRODUCT_ID = 0x8003
message = new byte[8]
hidServices = HidManager.getHidServices()
devices = hidServices.getAttachedHidDevices()
//hidServices.addHidServicesListener()
dymo = hidServices.getHidDevice(VENDOR_ID, PRODUCT_ID, null)
if (dymo){
	val = dymo.read(message, 1000)
}
hidServices.shutdown()
if (dymo){ 
	return message[0] + ' ' + message[1] + ' ' + message[2] + ' ' + message[3] + ' ' + message[4] + ' ' + message[5] + ' ' + message[6] + ' ' + message[7] 
} else {
	return 'not found'
} //end if

rwu2359, 2017-03-01:
strange how when you write out your pseudo code for someone else the answer appears..
I needed to add dymo.close() otherwise it stays open but does not read

the code for languages that can read the bytes to correctly return the ray weight is something like in Groovy
byte[] data = [ message[5], message[4] ]
	wrapped = ByteBuffer.wrap(data)
	raw_weight = wrapped.getShort()
	if ( message[2] == DATA_MODE_OUNCES ) { 
		scale = 10.power(message[3].toString().toShort())
		weight = raw_weight * scale
		if ( weight >= 16 ) { 
			LBS = (weight/16 ).toInteger()
			OZ = weight.mod(16).round(2)
			weight = LBS + 'lb ' + OZ + 'oz'
			//or
			//weight = weight + 'oz'
		} else { 
			weight = weight + 'oz'
		}
 	} else { 
		weight = String.format("%,d", raw_weight) + 'g'	
	}

gary-rowe, 2017-03-01:
Great that you've got it working, and thanks for the update. Sometimes [rubber ducking](https://en.wikipedia.org/wiki/Rubber_duck_debugging) is the best debugging tool.

closed 2017-03-01 by gary-rowe