gary-rowe/hid4java #53: String descriptors constantly accessed on Windows 7

vinbedded, 2017-04-20:
I developed a small app using hdi4java.  The app works great and as expected on Mac, but when I try the same jar file on Windows 7 64bit the string descriptors are constantly being read from the device.  This happens when I start the HID Services (i.e. hidServices = HidManager.getHidServices();), so the problem is occurring before connecting to any device.  I can see the string descriptors constantly going by on the USB analyzer.  
Has anyone seen this behavior before?  

Unfortunately I don't have access to a Windows 8/10 or Linux machine to see if the problem exists on other OS's.

Thanks,

gary-rowe, 2017-04-20:
This could be part of the auto-scan functionality that keeps track of which devices are currently attached to the machine.

vinbedded, 2017-04-20:
I tried changing the scan interval, but this had no effect.

vinbedded, 2017-04-21:
I forgot to mention that the constant access to the descriptors prevents the app from working and puts a heavy load on the computer.

vinbedded, 2017-04-22:
Confirmed same exact behavior on Windows 10.  Is there a fix for this?

gary-rowe, 2017-04-24:
@vinbedded Unfortunately I simply have no spare time to research what is causing this. Are you able to dig in and resolve the situation. A pull request can be verified and then added to the next release.

andyrozman, 2017-04-24:
@gary-rowe 
Didn't you say, sometime ago that there is a way to not use this auto-scan functionality? I know its not a fix, but it's more of a workaround.

gary-rowe, 2017-04-24:
@andyrozman Yes, it's a setting in the API that can be turned off via `setScanMode` in the `develop` branch. The `UsbHidDeviceExample` provides an example of its use.

gary-rowe, 2017-04-24:
(On a side note I'd love it if someone could review the code in some of the outstanding issues for correctness then I'd be able to progress version 0.5.0)

andyrozman, 2017-04-24:
@gary-rowe 
Could you point out which issues so that I perhaps (or someone else) could go through code?

gary-rowe, 2017-04-24:
@andyrozman Thank you! Here's the list: https://github.com/gary-rowe/hid4java/issues just locate the ones marked with the yellow "Awaiting Review" label

vinbedded, 2017-04-24:
@gary-rowe 
Unfortunately I was unable to resolve the situation.

gary-rowe, 2020-07-01:
Closing as it's been a long time and the latest develop-SNAPSHOT may have fixed it.

closed 2020-07-01 by gary-rowe