gary-rowe/hid4java #55: Unable to Open Device Handle

gnomff, 2017-05-11:
I am using a DS6878 Barcode Scanner (in HID mode) on OS X. I am running the example code which enumerates the devices allows me to see attached and detached events. However, `open` always returns false, and my OS retains ownership of the interface. `hidDevice.getLastErrorMessage()` returns ```Device not initialised```.

gary-rowe, 2017-05-15:
Just a quick check - is your device running as a keyboard/mouse in Windows? There is a note in the README covering this:

```
Check that the usage page is not 0x06 which is reserved for keyboards and mice. Windows opens these devices for its exclusive use and thus hid4java cannot establish its own connection to them. You will need to use the lower level usb4java library for this.
```

gnomff, 2017-05-17:
The usage page is 0x06, but I am not using Windows

gary-rowe, 2017-06-24:
Were you able to resolve this?

gary-rowe, 2020-07-01:
Closing as no response.

closed 2020-07-01 by gary-rowe