gary-rowe/hid4java #56: New version (0.5) and dependencies

andyrozman, 2017-06-22:
When is new release planned? 

Would it be possible to upgrade jna dependencies to latest/last 4.x (which would be 4.2.2)? I will do some tests over weekend if it works (at least on linux)...

Andy

gary-rowe, 2017-06-22:
Hi Andy,

I'm going to block out Saturday morning (UK time) to get a release done and
into Maven. Anything marked as "awaiting merge" will be in that release.

I'll look at updating the JNA dependencies since the project does need
dusting off and modernising.

Cheers,

Gary

andyrozman, 2017-06-22:
I think that changing jna to latest 4 shouldn't produce any special problems. It would be different if we would try to go with 5.x release, for that we would probably have to make few changes... 

Sorry to be bother, but I am using nrjavaserial for Serial communication in my project and new version has also dependency to jna (4.2.2), and using both libraries with different jna's would never work...

gary-rowe, 2017-06-24:
Ready for review and merge.

gary-rowe, 2017-06-24:
Merged and closed.

closed 2017-06-24 by gary-rowe