gary-rowe/hid4java #59: JVM crashed

lightway82, 2017-07-12:
Sometimes JVM is crashed after closing the device in Linux. In Windows it not appear.
I use Ubuntu 16.04 

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f1031548352, pid=5125, tid=0x00007f105a872700
#
# JRE version: Java(TM) SE Runtime Environment (8.0_112-b15) (build 1.8.0_112-b15)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.112-b15 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libusb-1.0.so.0+0x8352]  libusb_submit_transfer+0x142
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/anama/Projects/JavaUsb/hs_err_pid5125.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)`
```

Class for test it:
```

import org.hid4java.*;
import org.hid4java.event.HidServicesEvent;

/**
 * <p>Demonstrate the USB HID interface using a production Bitcoin Trezor</p>
 *
 * @since 0.0.1
 *
 */
public class TestUSB implements HidServicesListener {

    private static final short VENDOR_ID = (short) 0xfc58;
    private static final short PRODUCT_ID = (short)0x0001;
    private static final int PACKET_LENGTH = 64;
    public static final String SERIAL_NUMBER = null;

    public static void main(String[] args) throws HidException {

        TestUSB example = new TestUSB();
        example.executeExample();

    }

    public void executeExample() throws HidException {

        // Configure to use custom specification
        HidServicesSpecification hidServicesSpecification = new HidServicesSpecification();
        hidServicesSpecification.setAutoShutdown(true);
        hidServicesSpecification.setScanInterval(500);
        hidServicesSpecification.setPauseInterval(5000);
        hidServicesSpecification.setScanMode(ScanMode.SCAN_AT_FIXED_INTERVAL_WITH_PAUSE_AFTER_WRITE);

        // Get HID services using custom specification
        HidServices hidServices = HidManager.getHidServices(hidServicesSpecification);
        hidServices.addHidServicesListener(this);

        // Start the services
        System.out.println("Starting HID services.");
        hidServices.start();

        System.out.println("Enumerating attached devices...");

        // Provide a list of attached devices
        for (HidDevice hidDevice : hidServices.getAttachedHidDevices()) {
            System.out.println(hidDevice);
        }

        // Open the device device by Vendor ID and Product ID with wildcard serial number
        HidDevice hidDevice = null;
        for(int i=0;i<30;i++) {

            hidDevice = hidServices.getHidDevice(VENDOR_ID, PRODUCT_ID, SERIAL_NUMBER);
            if (hidDevice != null) {
                System.out.println(i);
                sendMessage(hidDevice);
            }
        }

        // Shut down and rely on auto-shutdown hook to clear HidApi resources
        hidServices.shutdown();

    }

    @Override
    public void hidDeviceAttached(HidServicesEvent event) {

        System.out.println("Device attached: " + event);

        // Add serial number when more than one device with the same
        // vendor ID and product ID will be present at the same time
        if (event.getHidDevice().isVidPidSerial(VENDOR_ID, PRODUCT_ID, null)) {
            sendMessage(event.getHidDevice());
        }

    }

    @Override
    public void hidDeviceDetached(HidServicesEvent event) {

        System.err.println("Device detached: " + event);

    }

    @Override
    public void hidFailure(HidServicesEvent event) {

        System.err.println("HID failure: " + event);

    }

    private void sendMessage(HidDevice hidDevice) {

        // Ensure device is open after an attach/detach event
        if (!hidDevice.isOpen()) {
            hidDevice.open();
        }

        hidDevice.close();

    }

}
```

I use cycle for catch problem, but jvm cached periodically  without open-close cycle

gary-rowe, 2017-07-12:
This might be due to an internal bug in the OpenJDK JVM but you are using the latest available version (u112) so I'm inclined to suspect a problem in the underlying libraries and JNA. 

Could you pick through the additional error log `/home/anama/Projects/JavaUsb/hs_err_pid5125.log` to see if there is more detail that you could post here?

lightway82, 2017-07-12:
[hs_err_pid5125.txt](https://github.com/gary-rowe/hid4java/files/1142464/hs_err_pid5125.txt)

Andrei-Pozolotin, 2018-06-02:
[hs_err_pid11468.log](https://github.com/gary-rowe/hid4java/files/2065066/hs_err_pid11468.log)

AdsonEsteves, 2018-08-15:
[hs_err_pid1394.log](https://github.com/gary-rowe/hid4java/files/2291217/hs_err_pid1394.log)
[hs_err_pid17555.log](https://github.com/gary-rowe/hid4java/files/2291218/hs_err_pid17555.log)

aviditynic, 2018-10-29:
Any progress on this item?  Similar behavior on Raspberry Pi (ARM) and Mac (iOS)

gary-rowe, 2019-10-29:
This could be related to a general requirement to upgrade `hidapi` so I'll reference it against #81 and ask anyone affected to try the new libraries currently on the `develop` branch.

Closing to avoid duplication.

closed 2019-10-29 by gary-rowe