gary-rowe/hid4java #60: Can't open devices in Windows 10 (new hidapi needed)

sean-horton, 2017-07-14:
It looks like the included hidapi dll is from 2 years ago which seems to be broken on Windows 10. There was a fix in hidapi 10 months ago which allows devices to be opened in Windows 10, issue seen here https://github.com/signal11/hidapi/issues/231

I've not tested the fix yet, but will give it a try over then next few days. Figured I would report it though.

gary-rowe, 2017-07-14:
Thanks for the update. The binaries haven't been changed for a long time so could certainly do with a refresh for 0.6.0.

gary-rowe, 2017-07-17:
OK, good to know. I'll see if I can rebuild my Windows compilation environments over the weekend and look into a freshening of the binaries for 0.6.0. 

If it's turning hairy I'll look to the community to offer up suitable binaries - I'd prefer deterministic builds with SHA256 hashes confirmed by other contributors for safety (that includes my own contributions too).

sean-horton, 2017-07-28:
Hi Gary, is there any update on this or ETA? Thanks.

gary-rowe, 2019-10-29:
Well it's been a *long* time, but I've finally be able to revisit the project. I've updated the binaries and they are currently in the `develop` branch. Since this is probably now a duplicate of #81 I'll mark this one as closed to avoid distractions.

closed 2019-10-29 by gary-rowe