gary-rowe/hid4java #66: Could you provide a basic GetDescriptor example?

badvision, 2017-10-02:
I've been able to hit Bluetooth game controllers with this library (extremely cool) but having no luck fetching the descriptors so I can write code that is able to adapt to the available features of the game controller in question.  Any pointers or a sample for this would be greatly appreciated.

badvision, 2017-10-02:
Please note, I am certainly not presuming you're going to provide a report parser, just an example of how to send the get_descriptor request and get the descriptor back as binary -- anything will help at this point as I am rather stuck.

Here's what I'm attempting after opening the device:
`        byte[] dataBuffer = new byte[64];
        Arrays.fill(dataBuffer, (byte) 0);
        int index = 0;
        dataBuffer[index++] = (byte) 0x080;
        dataBuffer[index++] = 6;
        dataBuffer[index++] = 0x021;
        dataBuffer[index++] = 0;
        dataBuffer[index++] = 1;
        dataBuffer[index++] = 0;
        dataBuffer[index++] = 64;
        dataBuffer[index++] = 0;

        int i = device.write(dataBuffer, (byte) 64, (byte) 0);
        System.out.printf("response: %d\n", i);
`
Response comes back as -1 and subsequent reads are empty until the joystick itself receives user input and it sends that as data packets.  So I get data, but unless I have a descriptor I have to write a one-off parser to read just that joystick's input.  This will, of course, not be useful to other people who have different types of game controllers.

gary-rowe, 2020-07-01:
Hi. Sorry for the slow response. Did you manage to get this sorted in the end?

gary-rowe, 2020-07-08:
No response, so closing.

closed 2020-07-08 by gary-rowe