gary-rowe/hid4java #67: Issues compiling Artifacts/.jar

DevZombi, 2017-10-24:
So I used the example project to create a simple command line application. I tried compiling that into an "artifact" with absolutely no success. I then downloaded a compiled version of the library from Maven central, created a new project and migrated my code to a new project, added the library .jar from maven central and I can compile a new artifact, but when I run it I get 
`
Exception in thread "main" org.hid4java.HidException: Hidapi did not initialise: com/sun/jna/Structure$ByReference`

I am not certain what I'm doing for it not to run outside of the example project. Any help would be amazing.

DevZombi, 2017-10-24:
BTW just wanted to add how great this project is. Simply an amazing undertaking, thanks to everyone that contributed.

gary-rowe, 2018-04-03:
Hi. Apologies for the slow response to this - I've been unable to work on this library for a while.

It sounds like you've got a dependency issue. Can you let me know if you've managed to resolved this yourself or need some more detailed guidance?

gary-rowe, 2019-10-29:
Closing since no further info.

closed 2019-10-29 by gary-rowe