gary-rowe/hid4java #68: Memory leak

NeacsuCristian, 2017-10-24:
Hello,

First I want to congratulate you for this awesome project.
Secondly, I want to report you a small bug (maybe a naive usage from my side of the package).
Let's assume we have have a dumb function like this:
public static void test(HidDevice usbDevice) {
    	usbDevice.open();
    	usbDevice.close();
}

If this will be called in a while(true) loop, after some hundreds of iterations will fail with 
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f9e35a7e75f, pid=26086, tid=0x00007f9e1572e700

I figure out this, due to the fact that every time I was sending a command through USB I was opening and closing the port. (not practical at all, but still it should work). If you open/close the port only once, the reading and writing commands are working flow-less.

Hopefully this observation might help in the future.
Cristian-Stefan

gary-rowe, 2018-04-03:
Hi! I'll be working on this library starting Mon 16th April in order to release version 0.6.0. I can take a look at this issue then.

darrazo, 2019-05-20:
Hi Gary, thanks a lot for the work on this great library! is the 0.6.0 release still planned?

gary-rowe, 2019-10-29:
@darrazo Yes, I've finally cleared my schedule enough to get back to work on this project. Lots of clearing up to do.

This might be related to #81 since it's a native issue. I'll treat it as a duplicate and handle issues there.

closed 2019-10-29 by gary-rowe