gary-rowe/hid4java #69: Version Number

Vzor-, 2017-10-27:
Hey, just wondering if you could toss a version number somewhere into the code. Since this is an io library it is nice to be able to programmatically check the version via an api call.

andyrozman, 2017-10-27:
We could probably generate it in maven, so that artifact version would be
visible in code. I do the same in my project.

Vzor-, 2017-10-28:
That would be fine, I prefer an in-code version but I'm not picky.

gary-rowe, 2018-04-03:
Hi! I'll be working on this library starting Mon 16th April in order to release version 0.6.0. I can take a look at this issue then.

gary-rowe, 2019-10-29:
Added support for this in `develop` branch. Scheduled for release in 0.6.0+.

To test programmatically:
```java
// Use default service settings for example
HidServices hidServices = HidManager.getHidServices();
hidServices.getVersion();
```
To test on `develop` branch:
```bash
mvn clean package
java -cp target/hid4java-develop-SNAPSHOT.jar org.hid4java.HidServices
```
Output should be `develop-SNAPSHOT` or `0.6.0` etc after release on `master`.

closed 2020-07-01 by gary-rowe