gary-rowe/hid4java #70: Listener detect new USB device connected [Java/Windows]

JPizarro14, 2018-02-15:
Hi!

I'm developing a new project and I need a listener to detect when a USB device has been connected or disconnected. I've been searching different solutions for Java on Windows but I didn't found anything. Is there any possibility to develop that listener using hid4java?

Thank you so much
Best regards

gary-rowe, 2018-04-03:
Hi! I'll be working on this library starting Mon 16th April in order to release version 0.6.0. I can take a look at this issue then.

JPizarro14, 2018-04-23:
Hi Gary! Thank you very much! Do you have release date for version 0.6.0?

Best regards

gary-rowe, 2020-07-01:
The library supports attach/detach events so I think I can close this issue.

closed 2020-07-01 by gary-rowe