gary-rowe/hid4java #71: Problem with response from device

MrHazee, 2018-03-26:
Hi there!

Im working on a project that involes Hid4java and a HID-device. Its working great in general but I have this little problem regarding the response time from my device. 
If I call hidDevice.read(data, 100): no respons.
If i call hidDevice.read(data, 1500): Response but to slow.
If i call just hidDevice.read(data): Extremely fast and nice response but im stuck in the function. Its blocked?!
Any suggestions to solve this matter? Probably not a biggie but I have not that much experience so.. would appreaciate any help!

Thanks /Rasmus

MrHazee, 2018-03-28:
Solved.

closed 2018-03-28 by MrHazee

andyrozman, 2018-03-29:
Can you share what you did in case, somebody else has the same problem?

MrHazee, 2018-03-29:
Actually Im not sure what caused the problem. But Im pretty sure it was not the API. Probably I messed the timing up between my custom tx/rx threads.
It now works flawless with hidDevice.read(data,2000); ie very fast respons.