gary-rowe/hid4java #72: Opening reopening devices

MrHazee, 2018-03-30:
Hi there!
First, thanks for this beautiful api. Great!!
But I have a couple of problems I just cant work out.

Win 10

First,
If I connect to my device (both physically and software-wise) and then unplug the USB cable, I get tons of messeges to the Console.
In my native language Swedish, it looks like this:
"Enheten ?r inte ansluten."
"Enheten ?r inte ansluten."
"Enheten ?r inte ansluten."
...times infinity..

... which translates to "The device is not connected". I´ve attached a hidDeviceDetached-listener
but despite that I get about 50 error messages before i can stop it.
It lies in the JNA so I have my hand tied up. Any suggestions how I can solv this matter?

Second,
I think I´ve read all the post about open a device, close a device, and the open it again. But I cant get it right.
No matter what, i get "Device has not been opened" the second time I try to open it. Right now it feels like an impossible task..
I would like to release the handle, like resetting the application, but only the hid-part of it.
I havent changed FILE_SHARE_READ | FILE_SHARE_WRITE yet, is that inside the .dll??
I´ve tried both t not make new instances of hidServices and to make new when I re-connect..
When Im hit my disconnect buttom I´ve tried this;
hidDevice.close.
hidServices.shutdown
HidApi.Exit
But nothing seems to work...
Hope anyone can help me on this, ripping my hair of here..
Thanks! /Rasmus

gary-rowe, 2018-04-03:
Hi! I'll be working on this library starting Mon 16th April in order to release version 0.6.0. I can take a look at this issue then.

MrHazee, 2019-01-23:
Hi Gary, Hope ur good! Just wondering if you have any new on this subject? 
Best regards /Rasmus

gary-rowe, 2020-07-01:
Hi. It's been a while but I'm finally able to get back to regularly working on this library. I've pushed some changes to develop-SNAPSHOT that may fix this. If you're able could you review?

gary-rowe, 2020-07-08:
Given that others have had success with recent changes and Windows I'll mark this as closed.

closed 2020-07-08 by gary-rowe

MrHazee, 2021-02-22:
Hi Gary,
Thanks for getting back to me and Im sorry I could not review, I was deep down in other projects at the time.