gary-rowe/hid4java #74: negative shorts

jonbanjo, 2018-10-01:
You’ve probably got these anyway if you are working on a new version but a couple of observations found while playing with a usb logging multimeter which has a product id of e008.

HidDevice.toString() prepends “ffff” to the product id.
HidDevice.isVidPidSerial() does not compare the int parameters with the device short product id correctly.

I’d guess there may be a couple of others where the short representation is negative but the library worked well otherwise for my brief experiment.

gary-rowe, 2018-10-01:
Thanks for bringing this to my attention. It may be a while before I can get around to fixing it.

gary-rowe, 2020-07-01:
Hi. It's been a long while, but I have managed to get around to this. 

Can I verify that your concern with the `short` comparison is the unsigned upcast? For example a structure similar to this:

```
int intVendorId == shortVendorId & 0xFFFF;
```

See https://github.com/gary-rowe/hid4java/blob/29a20a0484e3f2faa09935d4cb73c5536ed2d8a6/src/main/java/org/hid4java/HidDevice.java#L402

gary-rowe, 2020-07-01:
Closing as this is essentially a duplicate of #65

closed 2020-07-01 by gary-rowe