gary-rowe/hid4java #78: Reading from HID Keyboard returns mostly zeros.

strokine, 2019-06-20:
Hello, first of kudos for the lib. This is the only lib which worked for me. 
I'm connecting a QR Reader to Raspberry Pi and trying to read the code from it. 
This is a USB device which acts as a keyboard. If I open a text editor and scan a code, it just prints it there. So I know the device is working.
Now I want to get access to that output from my Java code.
I was able to find the device and read from it, but I'm getting a strange result:
```java
boolean moreData = true;
        while (moreData) {
            byte data[] = new byte[PACKET_LENGTH];
            int val = hidDevice.read(data);
            System.out.println("#### "+val);
            switch (val) {
                case -1:
                    System.out.println(hidDevice.getLastErrorMessage());
                    break;
                case 0:
                    System.out.println("no more data");
                    moreData = false;
                    break;
                default:
                    System.out.print("< [");
                    for (byte b : data) {
                        System.out.printf(" %02x", b);
                    }
                    System.out.println("]");
                    break;
            }
        }
```

This code prints:
```
#### 8
< [ 02 00 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
#### 8
< [ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
#### 8
< [ 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
#### 8
< [ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
```
As far as I can tell I need to work only with first 8 bytes of each response, but those bytes are mostly zeros. How I could get an actual charachter or string the device returns?

Here is the device info, just in case:
HidDevice [path=0001:0009:00, vendorId=0x1eab, productId=0xffff8203, serialNumber=null, releaseNumber=0x100, manufacturer=NewLand, product=HidKeyBoard, usagePage=0x0, usage=0x0, interfaceNumber=0]

Any help is appreciated,
Thx a lot

YukiDmZhao, 2019-08-19:
I'm in the same situation as you are. Have you solved it?

strokine, 2019-08-19:
I've ended up writing a custom more logic which would mimic the keyboard input reading. So I had to ignore all zeros, and converting the actual codes to ASCII, including the case than SHIFT is passed, to capitalize the character.
But the reading was so slow, I just switched the scanner to POS-HID mode and started using a standard USB reading library. It was faster and easier.

YukiDmZhao, 2019-08-21:
Thanks you give up this interface?

gary-rowe, 2019-10-29:
I've seen many problems with mice and keyboards using hidapi, so maybe it's best to use usb4java for those. Closing.

closed 2019-10-29 by gary-rowe