gary-rowe/hid4java #79: Critical MacOS fault

madhephaestus, 2019-06-22:
When using on MacOS 10.11.6, starting the device manager results in the JVM faulting and hard-crashing. 

Here is the Java frames from the error log:

```
Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  com.sun.jna.Native.invokePointer(JI[Ljava/lang/Object;)J+0
j  com.sun.jna.Function.invokePointer(I[Ljava/lang/Object;)Lcom/sun/jna/Pointer;+6
j  com.sun.jna.Function.invoke([Ljava/lang/Object;Ljava/lang/Class;Z)Ljava/lang/Object;+472
j  com.sun.jna.Function.invoke(Ljava/lang/reflect/Method;[Ljava/lang/Class;Ljava/lang/Class;[Ljava/lang/Object;Ljava/util/Map;)Ljava/lang/Object;+249
j  com.sun.jna.Library$Handler.invoke(Ljava/lang/Object;Ljava/lang/reflect/Method;[Ljava/lang/Object;)Ljava/lang/Object;+348
j  com.sun.proxy.$Proxy17.hid_enumerate(SS)Lorg/hid4java/jna/HidDeviceInfoStructure;+26
j  org.hid4java.jna.HidApi.enumerateDevices(II)Lorg/hid4java/jna/HidDeviceInfoStructure;+7
j  org.hid4java.HidDeviceManager.getAttachedHidDevices()Ljava/util/List;+10
j  org.hid4java.HidDeviceManager.scan()V+9
j  org.hid4java.HidDeviceManager$2.run()V+25
j  java.lang.Thread.run()V+11
v  ~StubRoutines::call_stub
[hs_err_pid1379.log](https://github.com/gary-rowe/hid4java/files/3317059/hs_err_pid1379.log)
```

madhephaestus, 2019-10-29:
Oracle Jdk 1.8, Mac OS 10 Sierra

gary-rowe, 2019-10-29:
Are you able to experiment with the new libraries in the `develop` branch?

madhephaestus, 2019-10-29:
i test by pulling from Maven in groovy, can you publish a 0.5.1 or some sort of pre-release on maven?

gary-rowe, 2019-10-29:
OK, I've pushed a version `develop-SNAPSHOT` to Maven Central. Want to try it?

npetillon, 2020-02-02:
Hi,
I've the same issue on MacOS 10.15. The last develop-SNAPSHOT don't fix the bug :/ 

`Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [CoreFoundation+0x47aaf]  CFSetGetValue+0x21
C  [CoreFoundation+0x47711]  __CFRunLoopFindMode+0xae
C  [CoreFoundation+0x5c739]  CFRunLoopAddSource+0x121
C  [IOKit+0x38803]  IOHIDDeviceScheduleWithRunLoop+0x4a
C  [IOKit+0x3c9a3]  __IOHIDManagerDeviceApplier+0x20f
C  [IOKit+0x54e4]  __IOHIDManagerDeviceAdded+0x338
C  [IOKit+0x5092]  __IOHIDManagerSetDeviceMatching+0x15b
C  [jna985620130375149263.tmp+0x1328]  hid_enumerate+0xc8
C  [jna5409389553399627327.tmp+0xdef4]  ffi_call_unix64+0x4c
C  0x00007000040d3408

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  com.sun.jna.Native.invokePointer(Lcom/sun/jna/Function;JI[Ljava/lang/Object;)J+0
j  com.sun.jna.Function.invokePointer(I[Ljava/lang/Object;)Lcom/sun/jna/Pointer;+7
j  com.sun.jna.Function.invoke([Ljava/lang/Object;Ljava/lang/Class;ZI)Ljava/lang/Object;+440
j  com.sun.jna.Function.invoke(Ljava/lang/reflect/Method;[Ljava/lang/Class;Ljava/lang/Class;[Ljava/lang/Object;Ljava/util/Map;)Ljava/lang/Object;+271
j  com.sun.jna.Library$Handler.invoke(Ljava/lang/Object;Ljava/lang/reflect/Method;[Ljava/lang/Object;)Ljava/lang/Object;+344
j  com.sun.proxy.$Proxy83.hid_enumerate(SS)Lorg/hid4java/jna/HidDeviceInfoStructure;+26
j  org.hid4java.jna.HidApi.enumerateDevices(II)Lorg/hid4java/jna/HidDeviceInfoStructure;+7
j  org.hid4java.HidDeviceManager.getAttachedHidDevices()Ljava/util/List;+10
j  org.hid4java.HidDeviceManager.scan()V+9
j  org.hid4java.HidDeviceManager$2.run()V+25
j  java.lang.Thread.run()V+11
v  ~StubRoutines::call_stub`

gary-rowe, 2020-07-01:
I've pushed another snapshot to Maven Central if you want to try again.

npetillon, 2020-07-01:
Thanks for your update.
I still have the same issue: 

```Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libobjc.A.dylib+0x6829]  objc_msgSend+0x29
C  [CoreFoundation+0x5c2a8]  CFRunLoopAddSource+0x121
C  [IOKit+0x393d3]  IOHIDDeviceScheduleWithRunLoop+0x4a
C  [IOKit+0x3d574]  __IOHIDManagerDeviceApplier+0x20f
C  [IOKit+0x5fce]  __IOHIDManagerDeviceAdded+0x301
C  [IOKit+0x5bd1]  __IOHIDManagerSetDeviceMatching+0x15b
C  [jna1930247527353171225.tmp+0x1328]  hid_enumerate+0xc8
C  [jna7199048041795482326.tmp+0xd94c]  ffi_call_unix64+0x4c
C  0x0000700004ab6550

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  com.sun.jna.Native.invokePointer(JI[Ljava/lang/Object;)J+0
j  com.sun.jna.Function.invokePointer(I[Ljava/lang/Object;)Lcom/sun/jna/Pointer;+6
j  com.sun.jna.Function.invoke([Ljava/lang/Object;Ljava/lang/Class;Z)Ljava/lang/Object;+486
j  com.sun.jna.Function.invoke(Ljava/lang/Class;[Ljava/lang/Object;Ljava/util/Map;)Ljava/lang/Object;+262
j  com.sun.jna.Library$Handler.invoke(Ljava/lang/Object;Ljava/lang/reflect/Method;[Ljava/lang/Object;)Ljava/lang/Object;+316
j  com.sun.proxy.$Proxy84.hid_enumerate(SS)Lorg/hid4java/jna/HidDeviceInfoStructure;+26
j  org.hid4java.jna.HidApi.enumerateDevices(II)Lorg/hid4java/jna/HidDeviceInfoStructure;+7
j  org.hid4java.HidDeviceManager.getAttachedHidDevices()Ljava/util/List;+10
j  org.hid4java.HidDeviceManager.scan()V+9
j  org.hid4java.HidDeviceManager$1.run()V+34
J 2503 C1 java.lang.Thread.run()V (17 bytes) @ 0x000000011b73810c [0x000000011b7380c0+0x4c]
v  ~StubRoutines::call_stub```

gary-rowe, 2020-07-01:
Hmm. That's the latest `hidapi` compiled under MacOS 10.15.5. Could you give me some more details about the device you're accessing and your initialisation code when starting `hid4java`?

Also I have a [Telegram group chat](https://t.me/joinchat/CtU4ZBltWCAFBAjwM5KLLw) if you want to get involved.

gary-rowe, 2020-07-01:
From the feedback on the Telegram group it appears that the latest develop-SNAPSHOT has resolved this issue.

closed 2020-07-01 by gary-rowe