gary-rowe/hid4java #82: Raspberry Pi scanning issue

madhephaestus, 2019-10-29:
Using the hid4java On windows, and x86_64 works well, but the same jar fails to open devices on a Raspberry Pi. Are there any special instructions or libraries that should be installed to make the Pi work the same as Ubuntu on x86?

Tested Non-working on
Pi 3
Pi 4 2gb
Pi 4 4gb
Pi Zero

Jvm: https://chriswhocodes.com/downloads/openjfx-8u60-sdk-overlay-linux-armv6hf.zip

gary-rowe, 2019-10-29:
Thanks for the testing - really appreciate the help.

The RPi support is a bit flaky at the moment. I'm having trouble cross-compiling the hidapi library to armv6. I'll keep at it and let you know here when the `develop` branch has an update.

madhephaestus, 2019-10-29:
I will offer, as a bounty, a free SmallKat https://hackaday.io/project/164727-smallkat-an-adorable-dynamics-oriented-robot-cat kit to whomever solves this issue :)

madhephaestus, 2019-10-30:
Upon further testing the HID device is found, and hid4java attempts to connect to it. The device receives no packets at all. The response times out after a minute. On timeout, device .disconnect is called, this causes a Fatal error:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x9e7e46e8, pid=1414, tid=0x9de91460
#
# JRE version: OpenJDK Runtime Environment (8.0_212-b01) (build 1.8.0_212-8u212-b01-1+rpi1-b01)
# Java VM: OpenJDK Client VM (25.212-b01 mixed mode linux-aarch32 )
# Problematic frame:
# C  [libusb-1.0.so.0+0x96e8]  libusb_get_next_timeout+0x60
#
# Core dump written. Default location: /home/pi/core or core.1414
#
# An error report file with more information is saved as:
# /home/pi/hs_err_pid1414.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

```

I have attached the core-dump
[hs_err_pid1414.log](https://github.com/gary-rowe/hid4java/files/3789869/hs_err_pid1414.log)

gary-rowe, 2019-10-30:
Thanks for the update. I'm still working on getting the RPi cross-compilation to work in the dockcross environment.

gary-rowe, 2019-10-30:
I've pushed a new version of `hid4java-develop-SNAPSHOT` to Maven Central. it should contain a working RPi shared library now.

madhephaestus, 2019-10-30:
YOU ROCK FOR DOING THIS SO FAST! One of my students is running through the tests tonight, ill report back in the morning :)

gary-rowe, 2019-10-30:
There'll be another update coming along in a moment. I've found a bug with the RPi library.

madhephaestus, 2019-11-05:
I cant seem to be able to load the snapshot. It isnt published here:

[https://repo1.maven.org/maven2/org/hid4java/hid4java/](https://repo1.maven.org/maven2/org/hid4java/hid4java/)

We are testing using the dep string:
```
implementation 'org.hid4java:hid4java:hid4java-develop-SNAPSHOT'
```
Can you tell me what dep string we should use to load this snapshot version? do i need to add a grab-resolver source?

madhephaestus, 2019-11-07:
Found how to include it, were still getting an error
```
org.hid4java.HidException: Hidapi did not initialise: Unable to load library 'hidapi': Native library (linux-arm/libhidapi.so) not found in resource path ([file:/home/pi/.groovy/grapes/org.hid4java/hid4java/jars/hid4java-develop-SNAPSHOT.jar, file:/home/pi/.groovy/grapes/net.java.dev.jna/jna/jars/jna-4.1.0.jar, file:/home/pi/.groovy/grapes/com.neuronrobotics/SimplePacketComsJava-HID/jars/SimplePacketComsJava-HID-0.13.1.jar, file:/home/pi/.groovy/grapes/com.neuronrobotics/SimplePacketComsJava/jars/SimplePacketComsJava-0.10.1.jar])
	at org.hid4java.HidDeviceManager.<init>(HidDeviceManager.java:89)
	at org.hid4java.HidServices.<init>(HidServices.java:88)
	at org.hid4java.HidServices.<init>(HidServices.java:76)
	at org.hid4java.HidServices.<init>(HidServices.java:63)
	at org.hid4java.HidManager.getHidServices(HidManager.java:50)
	at edu.wpi.SimplePacketComs.phy.HIDSimplePacketComs.connectDeviceImp(HIDSimplePacketComs.java:66)
```

OE4AMW, 2020-06-03:
Hi @madhephaestus,
I experienced the same issue (Unable to load library 'hidapi': Native library (linux-arm/libhidapi.so) not found in resource path) when using https://raw.githubusercontent.com/bitcoin-solutions/mbhd-maven/master/snapshots/org/hid4java/hid4java/develop-SNAPSHOT/hid4java-develop-SNAPSHOT.jar (as described in README.md)

For some reason, the hidjava-develop-SNAPSHOT.jar does not contain a folder linux-arm with a libhidapi.so, instead there is only a libhidapipi.so in the root-folder.
Manually "fixing" the jar (by renaming it to zip, changing the folder/filename, re-zipping and renaming back to .jar) helped in my case.

gary-rowe, 2020-07-08:
I've updated the README in `develop` branch to include suitable Gradle configuration for the snapshot releases.

```gradle

repositories {
    mavenCentral()
}

dependencies {
    implementation('org.hid4java:hid4java:develop-SNAPSHOT')
}

```

If this fixes your problems, please let me know.

gary-rowe, 2020-07-09:
I should add that Maven Central has a new develop-SNAPSHOT that should fix your issue.

gary-rowe, 2020-07-15:
I've verified that the latest build (using JNA 5.5.0) works on a RPi Model B Rev 2 which is `linux-arm` using hard float. This should sort out any issues with `SIGILL (0x4)` or `SIGSEGV (0xb)`.

Closing.

closed 2020-07-15 by gary-rowe