gary-rowe/hid4java #83: hid4java Android

Chakib-Temal, 2019-11-07:
Hi i try to implement this library in an Android OS

to add the depo maven in build.gradle (module:app)

dependencies {
    compile 'org.hid4java:hid4java:0.5.0'
}
when the execution is in this line 
        HidServices hidServices = HidManager.getHidServices(hidServicesSpecification);

i have an exception : 
Caused by: org.hid4java.HidException: Hidapi did not initialise: Native library (com/sun/jna/android-arm/libjnidispatch.so) not found in resource path (.)

have you a solution ? an idea ? thank u

gary-rowe, 2019-11-07:
At present Android isn't supported. I'm considering it for the latest 0.6.0 release which has some newer native libraries. 

If you'd like to try compiling `hidapi` native libraries for your platform there are enhanced instructions on the `develop` branch. Obviously, if you want it included in the final release, let me know how you get on here on this issue.

Chakib-Temal, 2019-11-07:
have you any idea ? how to start ?

gary-rowe, 2019-11-08:
I had a quick look at it earlier today using `dockcross`. Essentially you'll need to do the following:

1. Take a look at the `README.md` in the `develop` branch of this project under compiling for different platforms.
1. Examine the `build-hidapi.sh` for Android to see my (failed) attempts to get it working.
1. Install Docker, dockcross etc to get your virtual build environment up and running.
1. Attempt to cross-compile the hidapi native library for arm64. (You may need to fiddle with the `./configure --host=<something>` command to get it to work). I reached a barrier with `librt.so` being missing for the aarch64 compiler.

As you can see it's quite complex and you may find it easier to compile `hidapi` on an Android machine using a native compiler. See the `README.md` in this project for a link to the `libusb/hidapi` project so you can git clone it and follow its build instructions.

I may get some time next week to look at this but hopefully the above will get you started.

Chakib-Temal, 2019-11-15:
> I had a quick look at it earlier today using `dockcross`. Essentially you'll need to do the following:
> 
> 1. Take a look at the `README.md` in the `develop` branch of this project under compiling for different platforms.
> 2. Examine the `build-hidapi.sh` for Android to see my (failed) attempts to get it working.
> 3. Install Docker, dockcross etc to get your virtual build environment up and running.
> 4. Attempt to cross-compile the hidapi native library for arm64. (You may need to fiddle with the `./configure --host=<something>` command to get it to work). I reached a barrier with `librt.so` being missing for the aarch64 compiler.
> 
> As you can see it's quite complex and you may find it easier to compile `hidapi` on an Android machine using a native compiler. See the `README.md` in this project for a link to the `libusb/hidapi` project so you can git clone it and follow its build instructions.
> 
> I may get some time next week to look at this but hopefully the above will get you started.

it's seems hard to implement, i tried with the standard library android UsbManager
i send the message but the device didn't answer me

`result = connection.controlTransfer(0x21,0x09,0x0200,0x00, bytes, bytes.length,2000);`

it's seems the same function used in the C native library that you used for your library 

using your library , the parameters of this functions are writen automatically, now i don't know how to catch all this parameters from the log of C library (may be i can reproduce it with the android sdk standard) 

parameters are : 
requestType
request
value
index
buffer[] 
lenght of buffer 
timeout  

i don't really understand the 4 first parameters

how can i debog the C library to follow the logout of all parameters just before the send ?

Chakib-Temal, 2019-11-15:
it will be very nice if you can do the same thing for Android OS

gary-rowe, 2020-07-01:
I have some more time to work on hid4java these days so I'll see about getting Android working within the cross-compilation toolset.

Chakib-Temal, 2020-07-02:
> I have some more time to work on hid4java these days so I'll see about getting Android working within the cross-compilation toolset.

you don't need to develop anything on android

here is a class for HidDriver (example) with Android SDK
[HidDriverForAndroid](https://github.com/pazaan/600SeriesAndroidUploader/blob/master/app/src/main/java/info/nightscout/android/USB/UsbHidDriver.java)

gary-rowe, 2020-07-02:
Thanks, Chakib. That code looks ideal for Android developers to use and makes a lot more sense than having a dedicated HID library. I'll mark this as closed and remove Android support as a requirement.

closed 2020-07-02 by gary-rowe