gary-rowe/hid4java #84: Improper Function

goetzseb, 2019-11-28:
Hi all,

for my current project I was in search for a Java HID implementation and thus attracted by this project. Unfortunately I cannot get this to work somehow. I am able to list the devices and find the one I am targeting. But on the first write the HID library Returns with return Code -1 and the last error message states "Unzulässige Funktion" which I would translate to "Improper function".
I am on a Windows 64 bit machine and it tells me it is version 1903 (Build 18362.476). The current JDK I am using is Oracle 1.8.0_161 (32 bit). Previously I tested with a 64 bit JRE but the behavior is the same.

Is this a known issue or am I missing something?

gary-rowe, 2019-12-01:
I'm currently working towards a new 0.6.0. release. If you want to try the latest libraries please due the `develop-SNAPSHOT` version.

gary-rowe, 2020-07-01:
Closing due to no response.

closed 2020-07-01 by gary-rowe