gary-rowe/hid4java #88: The second connection is not working properly

WanManito, 2020-02-20:
In version 0.5.0, I tested read and write normally，
But here's what I did when I tried to free up resources
[image]

No exception was reported, but when the device was opened again, it was found to be null，as shown below ，
[image]

The normal situation would be as follows
[image]

Strangely enough, after I waited a while, the device was able to connect back to normal

I don't know if I release resources incorrectly or some other reason

gary-rowe, 2020-07-01:
That looks correct, but it may be that the resources are taking longer to be released than expected. Is it still happening with latest develop-SNAPSHOT version?

WanManito, 2020-07-06:
I dropped this solution in favor of USB4Java，I don't know if the solution you wrote supports Linux

gary-rowe, 2020-07-06:
Thanks for getting back to me. hid4java does support Linux but if you've moved on to use usb4java instead then I'll close this issue for now.

closed 2020-07-06 by gary-rowe