gary-rowe/hid4java #91: Error in Device.getIndexedString

recke96, 2020-05-19:
I'm trying to get a usb-device running in a java application which has only a proprietary windows/mac application. It reports itself as `[path=00xx:00xx:xx, vendorId=0x17b5, productId=0x10, serialNumber=000043C7, releaseNumber=0x106, manufacturer=Lunatone, product=MFT Sensor, usagePage=0x0, usage=0x0, interfaceNumber=0]`. 

Currently I'm just playing around since I don't know what I have to send to the device but `getIndexedString` seems to only return the first character.

* `getIndexedString(0x01)` returns `L` but capture shows `Lunatone`
* `getIndexedString(0x02)` returns `M` but capture shows `MFT Sensor`

also if the program continues running this produces a SIGSEV
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f535fad7048, pid=26846, tid=26880
#
# JRE version: OpenJDK Runtime Environment (13.0.2+8) (build 13.0.2+8)
# Java VM: OpenJDK 64-Bit Server VM (13.0.2+8, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# V  [libjvm.so+0xebb048]
```

gary-rowe, 2020-07-01:
Hi. Can you try out the latest develop-SNAPSHOT version to see if it fixes the issue?

recke96, 2020-07-02:
No the problem persists. 
Test program (kotlin):
```
fun main() {
    val service = HidManager.getHidServices()

   // service.start()

    val disc = service.getHidDevice(0x17b5, 0x10, null)
    if (disc != null) {
        val vendor = disc.getIndexedString(0x01)
        val product = disc.getIndexedString(0x02)

        println("Vendor: $vendor, Product: $product")

    } else {
        println( "Not attached" )
    }

   // service.shutdown()
}
```
Output shows that the error persists and some other error:
```
Vendor: L, Product: M
java: io.c:2146: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```
I'm on Linux 5.7.6-arch1-1 (ArchLinux) with Java version 
```
openjdk 14.0.1 2020-04-14
OpenJDK Runtime Environment (build 14.0.1+7)
OpenJDK 64-Bit Server VM (build 14.0.1+7, mixed mode)
```

recke96, 2020-07-02:
If I use the commented `start` `shutdown` calls the program crashes with SIGSEV:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f7de408e78b, pid=1135265, tid=1135294
#
# JRE version: OpenJDK Runtime Environment (14.0.1+7) (build 14.0.1+7)
# Java VM: OpenJDK 64-Bit Server VM (14.0.1+7, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libusb-1.0.so.0+0xb78b]
[error occurred during error reporting (printing problematic frame), id 0xb, SIGSEGV (0xb) at pc=0x00007f7e15e56b1d]

# Core dump will be written. Default location: Core dumps may be processed with "/usr/lib/systemd/systemd-coredump %P %u %g %s %t %c %h" (or dumping to /path/to/project/core.1135265)
#
# An error report file with more information is saved as:
# /path/to/project/hs_err_pid1135265.log
free(): invalid pointer
```

recke96, 2020-07-02:
this is using the develop-SNAPSHOT from https://raw.github.com/bitcoin-solutions/mbhd-maven/master/snapshots

recke96, 2020-07-02:
Using the `device.manufacturer` and `device.product` functions (converted to kotlin properties) yields the correct result but the other error is still there:
```
Vendor: Lunatone, Product: MFT Sensor
java: io.c:2146: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

recke96, 2020-07-02:
also `getIndexedString` in the hidapi for linux seems to return an error by default? [Linux hid.c](https://github.com/signal11/hidapi/blob/a6a622ffb680c55da0de787ff93b80280498330f/linux/hid.c#L794)

gary-rowe, 2020-07-02:
Hmm, that SIGSEV issue looks very similar to a different issue. Can you verify that you're using the latest `develop-SNAPSHOT` from Maven Central? It should be dated 1st July 2020 at the time of writing.

gary-rowe, 2020-07-02:
> also getIndexedString in the hidapi for linux seems to return an error by default? Linux hid.c

That code is in the Linux hid.c but the library for Linux is built using the [libusb/hid.c](https://github.com/signal11/hidapi/blob/a6a622ffb680c55da0de787ff93b80280498330f/libusb/hid.c#L1283) which does provide a return value. The cross-compiler environment fails if `libudev-dev libusb-1.0-0-dev` are not present.

recke96, 2020-07-02:
> this is using the develop-SNAPSHOT from https://raw.github.com/bitcoin-solutions/mbhd-maven/master/snapshots

I used this repository. There is no develop-SNAPSHOT in maven central?

gary-rowe, 2020-07-02:
Ah, you might not have enabled Maven snapshot versions in your project's pom.xml. Here's a link to the [relevant section in the README](https://github.com/gary-rowe/hid4java#i-want-to-try-the-latest-snapshot-release---i-know-the-risks).

recke96, 2020-07-02:
Do you know how to do this in gradle? Can't find the snapshots setting in the repository declarations

gary-rowe, 2020-07-08:
I've updated the README with Gradle configuration information. Can you let me know if this resolves the issue?

gary-rowe, 2020-07-15:
I think this is fixed, but I'm going to bump this into 0.7.0 release.

gary-rowe, 2020-08-20:
Given that 0.6.0 has been released and fixed a lot of SIGSEV issues I'm going to close this issue.

closed 2020-08-20 by gary-rowe