gary-rowe/hid4java #92: Unable to open device on windows

drome77, 2020-06-10:
Hey Gary, 
first of all thanks for you super work!
I'm working with a Boca printer on windows 10 64 were i installed jre 1.8.0_241 32bit, I'm able to find device but open function always return false. Device I'm trying to open is:
HidDevice [path=\\?\hid#vid_0a43&pid_0201#6&3a1cd3da&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}, vendorId=0xa43, productId=0x201, serialNumber=0000278143, releaseNumber=0x100, manufacturer=Boca Systems Inc., product=FGL46 Thermal Printer, usagePage=0x1, usage=0x0, interfaceNumber=-1]
as took your from the library, usagePage is 0x1 so it could be manage properly by your driver.
After open the getLastErrorMessage() return always Device not initialised, can you help me revolving it ?

drome77, 2020-06-11:
hey Gary, do you have any answer to my last comment ? do you plan to update the library on your next release ?

gary-rowe, 2020-06-11:
Hi. Apologies for disappearing on you. I'm currently very limited on the time I have available for hid4java but I'd like to get an updated release out. The closest I've got is in the develop branch, which you can compile for yourself (you'll get a version called "develop-SNAPSHOT") which is also in the Maven snapshot repository.

My current plan is to try to get a bit more community involvement by means of a Telegram group and a regular time slot where I can do some work on this. Currently this is every Wednesday at 5pm (UTC) onwards. Is this something that you'd be interested in?

drome77, 2020-06-12:
Sure , I'll try to give my help on this, I need an upgraded an stable release :)
Please take note that I have already tried the snapshot version from Maven repository but it doesn't solve my issue.
Please let me know what i have to do

gary-rowe, 2020-06-17:
I've set up a Telegram group to allow for real time communication. Please review the updated README for more information. I'll see about getting the new DLL into the project based on your findings.

Could you try building locally using your hidapi-0.9.0 and seeing if it works with your machine? I'd like to get a cross compilation environment working so that anyone can build it on any Docker enabled machine.

gary-rowe, 2020-07-01:
Given the recent work on the `develop` branch to cover this, would you say it's ready to close?

gary-rowe, 2020-07-01:
Confirmed working via [Telegram](https://t.me/joinchat/CtU4ZBltWCAFBAjwM5KLLw).

closed 2020-07-01 by gary-rowe