gary-rowe/hid4java #94: Inconsistent method names for adding / removing listener

MorpheusUK, 2020-07-16:
Hello,

Firstly thanks for producing this library it has made a project i am working on which is running on an older version of SLE Linux far easier. This is quite a trivial one but I notice that when i add the listener I do a addHidServicesListener(HidServicesListener listener). However to remover the listener the method seems to be unless i have misunderstood its purpose removeUsbServicesListener(HidServicesListener listener).

These should be named consistently.

Thanks

gary-rowe, 2020-07-18:
Agreed. I've made the change in the `develop` branch. Please take a look and let me know if it's OK.

MorpheusUK, 2020-07-20:
Yes that's great thanks.

gary-rowe, 2020-07-21:
Cool. Closing.

closed 2020-07-21 by gary-rowe