gary-rowe/hid4java #95: Add HidDeviceListener interface and implementation in order to listen to device events

mpetzold, 2020-07-21:
I would suggest to add a HidDeviceListener interface so that it should be possible to listen to device events.

gary-rowe, 2020-07-21:
Hi. This sounds similar in nature to #30, can you elaborate on what you have in mind? Perhaps a skeleton interface definition that would suit your project's needs?

gary-rowe, 2020-09-09:
Given the lack of activity I'm going to close this issue.

Laivindur, 2020-09-30:
Hi Gary,
Once we started to work with `hid4java` we have missed the feature is being suggested here. The same way `HidService` listen to `HidServiceEvents` for attachments and detachments, would be great if there were some way to attach our components to some listener to handle incoming messages from devices.
We have implemented this with a sort of `Observer` pattern wrapping the `hidDevice` instance we want to listen. It's a Runnable, running in a different thread which performs an endless loop around the statement 

```java
 while (active) {
            try {
                observe();
                //XXX CYV : Walkaround to avoid input messages to be read partialy due to interruptions
                //XXX CYV : Thread sleeps as much time as device keeps the input buffer open for new messages
                //XXX CYV : The lock and the sleep generates the innactivity period of readTimeoutx2
                Thread.sleep(readTimeout);
            }catch(InterruptedException ex) {
                LOG.error("USBHIDDeviceObserver' sleep interrupted!",ex);
                active = false;
            }
        }
```
And the observe
```java
 try {
            List<byte[]> output = new LinkedList<byte[]>();   
            boolean hasMoreData = Boolean.TRUE;
            do {
                byte[] data = new byte[maxPackageSize];
                // This method will now block for {readTimeout} ms or until data
                // is read
                int val = readDevice(data);
                switch (val) {
                case RUNTIME_ERROR:
                case TRANSFER_ERROR:
                    listener.onTransferError(new USBHidDeviceEvent(hidDevice, hidDevice.getLastErrorMessage()));
                    hasMoreData = Boolean.FALSE;
                    break;
                case END_OF_TRANSFER:
                    if (output.size() > 0) {
                        listener.onTransfer(
                                new USBHidDeviceEvent(hidDevice, new LinkedList<byte[]>(output)));
                        LOG.debug("observe() - " + output.size()+" packages ("+maxPackageSize+" length) were transfered from HidDevice " + hidDevice);
                    }
                    hasMoreData = Boolean.FALSE;
                    break;
                default:
                    // Fill up the outputstream
                    output.add(data);
                    break;
                }
            } while (hasMoreData);

        } catch (Exception ex) {
            LOG.warn("Error observing hid device data transfers. Cause: " + ex.getMessage(), ex);
            listener.onTransferError(new USBHidDeviceEvent(hidDevice, ex.getMessage()));
        } 
```
The listener interface is 

```java
public interface USBHidDeviceListener {

    /**
     * Handles errors during the loading of data comming from the HID Device.
     * {@link USBHidDeviceEvent} handled by {@code onTransferError} has no content nor error codes. Instead 
     * they provide with a {@code message} describing the error. 
     * @param usbHidDeviceEvent
     */
    void onTransferError(USBHidDeviceEvent usbHidDeviceEvent);

    /**
     * Handles the latest {@code content} transfered by the HID Device to the host. 
     * {@link USBHidDeviceEvent} handled by {@code onTransfer} has no output message nor response codes, just content.
     * @param usbHidDeviceEvent
     */
    void onTransfer(USBHidDeviceEvent usbHidDeviceEvent);

    /**
     * Handles a premature closed connection, making the device no longer available.
     * {@link USBHidDeviceEvent} handled by {@code onDeviceClose} has no output message nor content, just the serialNumber.
     * @param usbHidDeviceEvent
     */
    void onDeviceClose(USBHidDeviceEvent usbHidDeviceEvent);
}
```

The event

```java
public class USBHidDeviceEvent {

    //Null Object Pattern
    private static final String EMPTY_MESSAGE = "";
    
    private Optional<List<byte[]>> content;
    private Optional<String> message;    
    private final HidDevice hidDevice;
    
    public USBHidDeviceEvent(HidDevice hidDevice) {
        this.hidDevice = hidDevice;
        this.content = Optional.empty();
        this.message = Optional.empty();
    }
    
    public USBHidDeviceEvent(HidDevice hidDevice, String message) {
        this.hidDevice = hidDevice;
        this.content = Optional.empty();
        this.message = Optional.ofNullable(message);
    }
    
    public USBHidDeviceEvent(HidDevice hidDevice, List<byte[]> content) {
        this.hidDevice = hidDevice;
        this.content = Optional.ofNullable(content);
        this.message = Optional.empty();
    }
    
    public String getMessage() {
        return message.orElse(EMPTY_MESSAGE);
    }
    
    public List<byte[]> getContent() {
        return content.orElse(Collections.emptyList());
    }
    
    @Override
    public String toString() {
        return getClass().getSimpleName() + 
                "[Hid ="+hidDevice+"]"
                +"[Message="+getMessage()+"]";
    }    
}
```

I could have added `getHidDevice()` to the event, but I didn't because I don't want everybody being aware of the hid4java lib since it's an implementation detail.

Hope it helps!

gary-rowe, 2020-09-30:
Thanks for stepping up here @Laivindur. I've reopened the issue and I'll add in data read support as an event.

gary-rowe, 2020-10-01:
I've added some code to support this into the `issue-95` branch. If you'd like to take an early look and offer up some feedback on its usefulness it would help greatly.

Laivindur, 2020-10-01:
Hi Gary.

Unfortunately, we are kinda busy at the moment due to the release. We are performing several task forces to round some corners of the actual version. I will look at it as soon as I can.

gary-rowe, 2020-10-01:
No worries. I've still got a few extra items to add in but thought I'd let you know that work is underway.

Laivindur, 2020-10-09:
Hi Gary, I didn't forget this "feature" I'm still striving to find a gap in my schedule to look at it with care and the due attention. We have stumbled upon more critical problems and we thing hid4java is involved. I will open another issue and see what can we do.

gary-rowe, 2020-11-18:
Merged and closed.

closed 2020-11-18 by gary-rowe