gary-rowe/hid4java #96: Version 0.6.0 missing linux x86 hidapi library

MorpheusUK, 2020-07-22:
Hello,

Well the title says it all really the src/main/resources/linux-x86/libhidapi.so is missing from the 0.6.0 release. You may have already noticed this as I see a checkin related to cross compilation. This results in an error if attempting to run on a 32 bit linux system when the jna library tries to access it.

A quick solution for anyone who might need it is to download the x86 version of the library from the current trunk and then place it in the correct location before rebuilding the source. I haven't been able to test that much further as at that point I found that one of the system libraries was now too old (SLES 11 SP3) and for what I want to do at present 0.5.0 seems to be working fine and more than likely the actual device will be deployed on SLES 12 so trying to resolve this wasn't worth the time it would take..

Thanks,

Morpheus

gary-rowe, 2020-07-22:
Yeah, I think that got left out as it was quite old but I've added it back into `develop` so it'll be available in `0.7.0`. In the meantime, using `develop-SNAPSHOT` should be a drop-in fix. I've  pushed the most recent changes to Maven Central snapshots so hopefully that'll help.

gary-rowe, 2020-07-25:
Can you let me know if the `develop-SNAPSHOT` version fixed the missing library for you so I can close this issue?

MorpheusUK, 2020-07-27:
Unfortunately this is a work rather than personal project and priorities have shifted meaning I am not going to get a chance to try that again for a while. Plus as i said above when i downloaded the library from the develop branch and integrated it into the expanded 0.6.0 source zip and rebuilt, it then detected the library but the machine I was testing on had an older version of libudev.so if I recall and so it wouldn't initialise on that machine with the newer library (alas not a machine I fully control so even if I can find a suitably updated version I would have to install it on another machine without the actual device i am working on).

The commit comment and the fact that the library was in the current develop branch was why i suspected you might have been aware it was missing in the 0.6.0 release I just wasn't sure and if you hadn't noticed I didn't know if you would then have wanted to release a 0.6.1, leave it to the next release or make a note on the page etc.. Given the above if you want to close this you can as i am reasonably confident the issue has been resolved in 0.7.0  otherwise there will be a wait for me to do some rudimentary confirmation as time allows.

Sorry but looking at the current tasks on the board it may be over a week before i get back to this again.

gary-rowe, 2020-07-27:
No worries. Thanks for taking the time to report back. I'll mark this issue as closed and the `linux-x86` library will go out in 0.7.0.

closed 2020-07-27 by gary-rowe