gary-rowe/hid4java #98: Returning value is always -1 from getFeatureReport function

boraciner, 2020-08-19:
Hi! I'm trying to call getFeatureReport function but it's always returning -1. 

`hidApiLibrary.hid_get_feature_report(device.ptr(), report, data.length + 1);` is also returns -1 . what can be the problem?
Thanks.

gary-rowe, 2020-08-19:
Hi. If a feature report request is returning -1 then it indicates an error condition. You may find that calling `getLastErrorMessage()` will  provide more information on what has gone wrong.

boraciner, 2020-08-20:
it says `Incorrect function` =( I don't know why. =(

`byte[] mybytearray = new byte[512];
int retVal = myJoystick.getFeatureReport(mybytearray, (byte) 0x00);
if(retVal == -1) {
  System.out.println(myJoystick.getLastErrorMessage());  //Incorrect function
}`

gary-rowe, 2020-08-20:
Hi @boraciner. I did a quick bit of digging around to find a data sheet on the Speedlink Competition Pro joystick but it wasn't immediately available. Looking at the controller I'm not convinced that it is sophisticated enough to actually have feature reports which is why you're getting the error message.

I think your next step should be to find the data sheet on the internal USB controller for the device and discover the correct commands to detect its state (button up/down, left/right etc).

Also, you may find this project helpful: https://github.com/spurious/SDL-mirror (there is a lot of low-level HID work going on for a wide range of joysticks).

boraciner, 2020-08-20:
Thanks a lot for your quick response, I already extracted meaningful bytes and I'm handling these. I just wanted to have a report in order to use any kind of joystick independently from product vendor-specific implementation :) Have a nice day!

closed 2020-08-20 by boraciner