gary-rowe/hid4java #104: Error SIGSEGV (0xb) at attaching/detaching usb devices

Laivindur, 2020-10-09:
First of all, I'm unsure about how to report this bug and if it's really related to hid4java. The reasons for me to believe it are founded in some tests, core dumps analysis and different trials but, despite all the years working with Java, I have to admit that errors at such low level are very cryptic to me and I don't have a deep understanding of the JVM internals.

**Bug Description**
In runtime, when any USB device is connected, the JVM crashes due to SIGSEGV

```txt
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f0076fa09fd, pid=7111, tid=0x00007eff5eeb1700
#
# JRE version: OpenJDK Runtime Environment (8.0_265-b01) (build 1.8.0_265-8u265-b01-0ubuntu2~18.04-b01)
# Java VM: OpenJDK 64-Bit Server VM (25.265-b01 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libc.so.6+0x979fd]  cfree+0x3d
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread is native thread

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x0000000000000001
```
It happens in runtime when the app (Spring Boot app) already booted. Hid4JavaService is configured and started.
By this time
- If any USB already attached is detached, JVM crashes
- if any USB not attached is attached,  JVM crashes

This error never happens when hid4java is the only lib handling and managing USB devices. We have realised that the problem arises when it works alongside with usb4java.

How do we know?
We realised that, if we disable hid4java service initialization or usb4java initialization in the production code, the USB events are handled by the remaining API with no errors. The JVM doesn't crash. 

When both are up, there's a conflict at handling the USB events. Perhaps a race condition between both threads (service workers).

**To Reproduce**
It's hard to say because it's happening in a very complex application, which is multi-thread and on top of this, there's SpringBoot.
However, I have reproduced a very similar one, isolating both libs from Spring and the rest of the application. 

For brevity, I will share my `pom.xml` for you to know what is involved and the respective versions of the libs.

```xml
<project xmlns="http://maven.apache.org/POM/4.0.0"
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
	xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
	<modelVersion>4.0.0</modelVersion>

	<groupId>net.opentrends.abaxis</groupId>
	<artifactId>lab.ua</artifactId>
	<version>0.0.1-SNAPSHOT</version>
	<packaging>jar</packaging>

	<name>lab.ua</name>
	<url>http://maven.apache.org</url>

	<properties>
		<project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>
		<maven.compiler.source>1.8</maven.compiler.source>
		<maven.compiler.target>${maven.compiler.source}</maven.compiler.target>
		<junit.jupiter.version>5.6.2</junit.jupiter.version>
	</properties>

	<dependencies>
		<!-- USB4JAVA - http://usb4java.org/ -->
		<!-- Based on http://javax-usb.sourceforge.net/ -->
[... 330 lines cut ...]
	public static final String ANSI_RESET = "\u001B[0m";
	public static final String ANSI_BLACK = "\u001B[30m";
	public static final String ANSI_RED = "\u001B[31m";
	public static final String ANSI_GREEN = "\u001B[32m";
	public static final String ANSI_YELLOW = "\u001B[33m";
	public static final String ANSI_BLUE = "\u001B[34m";
	public static final String ANSI_PURPLE = "\u001B[35m";
	public static final String ANSI_CYAN = "\u001B[36m";
	public static final String ANSI_WHITE = "\u001B[37m";
}
```
As you see, it's a copy of your original `UsbEnumerationExample` i have just added the code to initialize `usb4java`. 
During the waiting time, if we plug any USB device, f(or example, I have used optical mice) it crashes 

```txt
Assertion 'device' failed at ../src/libsystemd/sd-device/sd-device.c:817, function sd_device_get_devtype(). Aborting.
```
**Desktop (please complete the following information):**
 - OS: Linux Ubuntu 16, 18, 20
 - Browser: NA
 - Version: NA
 - JRE: Oracle 1.8

**Additional context**
Add any other context about the problem here.

gary-rowe, 2020-10-10:
Usually a SIGSEV 0xb error is associated with a low level driver conflict between the JVM and supporting native libraries. 

Given

> We realised that, if we disable hid4java service initialization or usb4java initialization in the production code, the USB events are handled by the remaining API with no errors. The JVM doesn't [...]

I'm inclined to think in this case that it's `hidapi` and `libusb` attempting to gain control over the USB interface and conflicting with each other. The question how to fix it.

One approach is to minimise the interaction. Perhaps having `usb4java` handling attach/detach events and then `hid4java` only opening specific devices that are classed as HID. You can turn off attach/detach event polling using `ScanMode.NO_SCAN` in the `HidSpecification`.

If a deeper dig is required, you probably want to [review this answer on StackOverflow](https://stackoverflow.com/a/36118121/396747).

gary-rowe, 2020-10-26:
Any further info on this issue?

Laivindur, 2020-10-26:
Hi Gary.

Yes, but I'm afraid of not to be capable of giving a detailed explanation of the fix. 

We assumed what you suggested. Both native libs were competing since both were handling the USB attachments and detachments events. Maybe it was a lock, usb4java locking or reserving specific memory positions and hid4java causing the error at accessing to those sections or vice-versa. 

For some obscure reason, it stopped to fail when we modified certain routines of the project' start up. Basically, we moved some Spring configuration code to a different stage of the bootstrapping and after the change, the issue was gone. 

It's really rare but, to be honest, we have not delved into the problem. That said I think you were right on your assumptions.

That said, it makes me think about something I told early about the possibility to make both Java Libs to work together within the same project. Maybe it's not a good idea.

gary-rowe, 2020-10-27:
Thanks @Laivindur. Given that this was resolved in your project (albeit with mysterious Spring magic) I'll mark it as closed. I've added more info to the wiki in the troubleshooting section to assist others in their application design.

See https://github.com/gary-rowe/hid4java/wiki/Troubleshooting#i-get-a-sigsegv-0xb-when-starting-up for details.

closed 2020-10-27 by gary-rowe